project-chip/connectedhomeip #37551: [BUG]  In TC-SC-4.3 python script 'is_commissioning="true"' has to be added in test step1.

Ashwinigrl, 2025-02-13:
### Reproduction steps

In **TC-SC-4.3**, Test Step 1 states that the **"DUT must be commissioned on the same fabric as TH"**. However, the Python script missed add the **is_commissioning = "true"** variable. As a result, when validating with the UI, the test case appears under the **"No Commissioning"** section in the Python Testing Suite and fails to execute the steps due to the missing commissioning. Therefore, the **is_commissioning = "true"** variable must be added to the script.

[image]

[image]

**Reference Logs:**

[TC-SC-4.3(UI).log](https://github.com/user-attachments/files/18780227/TC-SC-4.3.UI.log)

### Bug prevalence

Whenver I do

### GitHub hash of the SDK that was being used

7d67dc904758bb68989596f401a843a31e81911d

### Platform

raspi

raul-marquez-csa, 2025-02-13:
PR
[Adds is_commissioning variable set to True #37566
](https://github.com/project-chip/connectedhomeip/pull/37566)

Ashwinigrl, 2025-05-14:
Issue resolved in **TH Version: v2.13-beta3.3+summer2025 Sha: de5f70f**. Hence closing it.