project-chip/connectedhomeip #37557: [BUG] [TC-TSTAT-2.2] python script is failing at test step 3c with constraint error response while running using the v2.12-beta2+spring2025 TH image sample app(thermostat app)

KishokG, 2025-02-13:
### Reproduction steps

1. Start the docker by sending the command: `docker run -v /var/credentials:/credentials -v /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket -v mapped_data_model:/root/python_testing/data_model -v /home/ubuntu/certification-tool/backend/test_collections/matter/sdk_tests/sdk_checkout/python_testing:/root/python_testing -v $(pwd):/launch_dir --privileged --network host -it connectedhomeip/chip-cert-bins:7d67dc904758bb68989596f401a843a31e81911d`
2. Please navigate to this path to access the scripts: `cd /root/python_testing/scripts/sdk `
3. In DUT side navigate to apps folder: `cd apps`
4. Advertise the DUT: `./thermostat-app`
5. Run the python script: `python3 TC_TSTAT_2_2.py --commissioning-method on-network --discriminator 3840 --passcode 20202021  --storage-path admin_storage.json  --PICS <PICS file path> --endpoint 1 `

### Bug prevalence

Whenever I try to run the TC-TSTAT-2.2 python script

### GitHub hash of the SDK that was being used

7d67dc904758bb68989596f401a843a31e81911d(v1.4-branch)

### Platform

raspi

### Platform Version(s)

RPI 4 with 8GB RAM

### Anything else?

**Scenario 1:** 

- while running the TC-TSTAT-2.2 python script with using the v2.12-beta2+spring2025 TH image sample app(thermostat app), the python script is failing at step 3c with constraint error response.

**Error log files:** 

DUT:  [TSTAT_2.2(DUTerror logs).txt](https://github.com/user-attachments/files/18784843/TSTAT_2.2.DUTerror.logs.txt)
TH: [TSTAT_2.2(TH error logs).txt](https://github.com/user-attachments/files/18784849/TSTAT_2.2.TH.error.logs.txt)

**Scenario 2:**

While running the TC-TSTAT-2.2 manually using the same sample app and chip-tool, the step 3c is passing without any issues.

**Error log files(chip-tool):** [TSTAT_2.2(manual).txt](https://github.com/user-attachments/files/18784854/TSTAT_2.2.manual.txt)
 

**Scenario 3:**

while running the TC-TSTAT-2.2 python script with using the sample app(thermostat app)`(Commit: 11a4c4a1f869494cb8569c7bd450a8319c6b0beb)`, the whole python script is passing without any issues.

**Error log files:** 
DUT:  [TSTAT_2.2(DUT PASS logs).txt](https://github.com/user-attachments/files/18784858/TSTAT_2.2.DUT.PASS.logs.txt)
TH:  [TSTAT_2.2(TH PASS logs).txt](https://github.com/user-attachments/files/18784860/TSTAT_2.2.TH.PASS.logs.txt)

nivi-apple, 2025-03-18:
Assigning to Hasty as he has more context than me about this test

KishokG, 2025-05-05:
Issue fixed in TH Version: v2.13-beta3+summer2025

closed 2025-05-05 by KishokG