project-chip/connectedhomeip #37748: [1.3]

lboue, 2025-02-24:
### Reproduction steps

The issue was introduced by #37640 because there are missing references:
- undefined reference to `emberAfOperationalCredentialsClusterSetVidVerificationStatementCallback
- undefined reference to `emberAfOperationalCredentialsClusterSignVidVerificationRequestCallback

I can't find those methods in [operational-credentials-server.cpp](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp)

**Error**
```shell
INFO    FAILED: chip-camera-app 
INFO    g++ -O0 -fPIC -Werror -Wl,--fatal-warnings -Wl,-z,defs -fdiagnostics-color -Wl,--gc-sections -pie @./chip-camera-app.rsp -o ./chip-camera-app
INFO    /usr/bin/ld: obj/gen/third_party/connectedhomeip/examples/camera-app/camera-common/zapgen/zap-generated/camera-common_zapgen.IMClusterCommandHandler.cpp.o: in function `chip::app::Clusters::OperationalCredentials::DispatchServerCommand(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&)':
INFO    /__w/connectedhomeip/connectedhomeip/out/linux-x64-camera/gen/third_party/connectedhomeip/examples/camera-app/camera-common/zapgen/zap-generated/IMClusterCommandHandler.cpp:447:(.text._ZN4chip3app8Clusters22OperationalCredentials21DispatchServerCommandEPNS0_14CommandHandlerERKNS0_19ConcreteCommandPathERNS_3TLV9TLVReaderE+0x649): undefined reference to `emberAfOperationalCredentialsClusterSetVidVerificationStatementCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::OperationalCredentials::Commands::SetVidVerificationStatement::DecodableType const&)'
INFO    /usr/bin/ld: /__w/connectedhomeip/connectedhomeip/out/linux-x64-camera/gen/third_party/connectedhomeip/examples/camera-app/camera-common/zapgen/zap-generated/IMClusterCommandHandler.cpp:456:(.text._ZN4chip3app8Clusters22OperationalCredentials21DispatchServerCommandEPNS0_14CommandHandlerERKNS0_19ConcreteCommandPathERNS_3TLV9TLVReaderE+0x6f6): undefined reference to `emberAfOperationalCredentialsClusterSignVidVerificationRequestCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::OperationalCredentials::Commands::SignVidVerificationRequest::DecodableType const&)'
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

4e25c49de1975e51100c7135323a1308859bb732

### Platform

android, other

### Type

Common Cluster Logic

bzbarsky-apple, 2025-02-25:
@lboue Those commands are not implemented yet, and not enabled in any .zap files.  It sounds like you enabled them without them being implemented?

lboue, 2025-02-25:
I wanted to activate them because I had this message, but you're right:
```
Application is failing the Cluster Specification as follows: 

  - ⚠ Check Cluster Compliance on endpoint: 0, cluster: Operational Credentials server, mandatory command: SetVidVerificationStatement incoming needs to be enabled
  - ⚠ Check Cluster Compliance on endpoint: 0, cluster: Operational Credentials server, mandatory command: SignVidVerificationRequest incoming needs to be enabled
  - ⚠ Check Cluster Compliance on endpoint: 0, cluster: Operational Credentials server, mandatory command: SignVidVerificationResponse outgoing needs to be enabled
```

closed 2025-02-25 by lboue

bzbarsky-apple, 2025-02-25:
> I wanted to activate them because I had this message

ZAP needs to be fixed to understand the concept of "provisional"....  Please file an issue on them @lboue ?

PR #38118 (2025-03-25): [VID Verification] Add empty handlers to fix ZAP issue
#### Changes

- Add empty unsupported handlers for the `SetVIDVerificationStatement` and `SignVIDVerificationRequest` commands, so that builds don't break if included in ZAP.
- Add SetVIDVerificationStatement and SetVIDVerificationStatement to all-clusters
  app.
- Add a temporary test for the fact that it's unimplemented.
- Fixes #37748

#### Testing

- New tests pass
commit b7899bce (2025-03-25): [VID Verification] Add empty handlers to fix ZAP issue (#38118)

* Update VidVerification->VIDVerification in ZAP XML

* ZAP Regen

* Run alchemy

* Regen zap

* Add unimplemented versionf of VID Verification commands

- Add SetVIDVerificationStatement and SetVIDVerificationStatement to all-clusters
  app.
- Make the cluster respond un-implemented.
- Add a temporary test for the fact that it's unimplemented.
- Fixes #37748

* Restyled by clang-format

* Fix integration test

---------