project-chip/connectedhomeip #37751: Documentation for python API

cecille, 2025-02-24:
It would be great if we could keep the documentation up to date automatically. There are a few things here
- ensure the documentation for the APIs is correct
  + special note - should add documentation around the SendCommand API that indicates that no return is success and it will throw on an error, which is different than the Read APIs

- Add CI to ensure the SDK documentation site is up to date
  + note - took a stab at this here - https://github.com/project-chip/connectedhomeip/pull/34761, wasn't able to get the paths correct, but it's a start.

juandediosg, 2025-03-26:
Created a new branch on my fork: "PyApi_Docs_UptoDate"

juandediosg, 2025-03-26:
For the first goal, all methods within classes (ChipDeviceControllerBase and ChipDeviceController) will be analyzed to ensure they have proper docstrings, focusing on raised exceptions and return values. Any methods lacking documentation will be updated to ensure clarity and consistency in our functions.

Once the API documentation is confirmed as Ok, the approach for the second point will be addressed.

juandediosg, 2025-04-01:
**PR as draft has been created:** https://github.com/project-chip/connectedhomeip/pull/38200

**Scope:** The task focusing on the `ChipDeviceController` class and its methods. The [Coding Style guide](https://project-chip.github.io/connectedhomeip-doc/style/CODING_STYLE_GUIDE.html#python) does not specify a specific docstring template, but it is understood that each method should include a general description along with sections for `raises` and `returns` when applicable. It has been noted that some methods lack information about their `return` or `raise` values in the docstrings, and this will be addressed as part of the task. 

Please find below the scope on the `ChipDeviceController` class and its methods that should be updated.
_**Path:** src/controller/python/chip/ChipDeviceCtrl.py_

| Function Name                                          | Action                             |
|--------------------------------------------------------|------------------------------------|
| CommissionThread                                        | Update docstring raise             |
| CommissionWiFi                                          | Update docstring return            |
| SetWiFiCredentials                                     | Update docstring raise             |
| SetThreadOperationalDataset                            | Update docstring raise             |
| ResetCommissioningParameters                           | Update docstring raise             |
| SetTimeZone                                            | Update docstring raise             |
| SetDSTOffset                                           | Update docstring raise             |
| SetTCAcknowledgements                                   | Update docstring raise             |
| SetSkipCommissioningComplete                           | Update docstring raise             |
| SetDefaultNTP                                          | Update docstring raise             |
| SetTrustedTimeSource                                   | Update docstring raise             |
| SetCheckMatchingFabric                                 | Update docstring raise             |
| GenerateICDRegistrationParameters                      | Update docstring return            |
| EnableICDRegistration                                  | Update docstring raise             |
| DisableICDRegistration                                 | Update docstring raise             |
| GetFabricCheckResult                                   | Update docstring return            |
| get_rcac                                               | Update docstring return            |
| NOCChainCallback                                       | Update docstring return            |
| IssueNOCChain                                          | Update docstring return            |
| SetDACRevocationSetPath                                | Update docstring raise             |
| Shutdown                                              | Update docstring raise, return    |
| CheckIsActive                                          | Update docstring raise             |
| IsConnected                                            | Update docstring return            |
| UnpairDevice                                           | Update docstring return            |
| CloseBLEConnection                                     | Update docstring raise             |
| ExpireSessions                                         | Update docstring raise             |
| GetTestCommissionerUsed                                | Update docstring return            |
| SetTestCommissionerSimulateFailureOnStage              | Update docstring return            |
| SetTestCommissionerSimulateFailureOnReport             | Update docstring return            |
| SetTestCommissionerPrematureCompleteAfter              | Update docstring return            |
| CheckTestCommissionerCallbacks                         | Update docstring return            |
| CheckStageSuccessful                                   | Update docstring return            |
| CheckTestCommissionerPaseConnection                    | Update docstring return            |
| GetAddressAndPort                                      | Update docstring return            |
| DiscoverCommissionableNodes                            | Update docstring return            |
| GetDiscoveredDevices                                   | Update docstring return            |
| GetIPForDiscoveredDevice                               | Update docstring return            |
| OpenCommissioningWindow                                | Update docstring return            |
| GetCompressedFabricId                                  | Update docstring raise, return    |
| GetFabricIdInternal                                    | Update docstring raise, return    |
| GetFabricIndexInternal                                 | Update docstring raise, return    |
| GetNodeIdInternal                                      | Update docstring raise, return    |
| GetClusterHandler                                      | Update docstring return            |
| FindOrEstablishPASESession                             | Update docstring return            |
| GetConnectedDeviceSync                                 | Update docstring raise             |
| ComputeRoundTripTimeout                                | Update docstring return            |
| GetRemoteSessionParameters                             | Update docstring raise, return    |
[... cut ...]
| ReadAttribute                                          | Check code raise by docstring      |
| ReadEvent                                              | Check code raise by docstring      |
| SetIpk                                                 | Update docstring raise             |
| InitGroupTestingData                                   | Update docstring raise             |
| CreateManualCode                                       | Update docstring raise, return    |

_**Note:** _Private methods out of scope__

khodya, 2025-04-01:
https://peps.python.org/pep-0257/

juandediosg, 2025-04-03:
New push on the PR: Started updating docstrings for functions and methods. Still in draft mode

juandediosg, 2025-04-07:
Working on implement the draft PR **ChipDeviceCtrl API documentation in build #34761** (with `pydoc`), trying to get it working locally to create the PR. 
Once functional, I will explore integrating this functionality into the CI jobs.

juandediosg, 2025-04-09:
For Phase 1 PR: [Documentation for python API - Docstring Updates #38200](https://github.com/project-chip/connectedhomeip/pull/38200):

- I have added my replies on comments. 
- Current status, Gteam approval done, jJust waiting for the second company's approval.

juandediosg, 2025-04-16:
**First stage** [Documentation for python API - Docstring Updates #38200](https://github.com/project-chip/connectedhomeip/pull/38200): The PR has been merged

juandediosg, 2025-06-03:
### Current status on Phase 3

The Phase 3 PR Integration of the documentation generation into the CI pipeline (to be done in a future PR) has been completed and implemented on the same PR as Phase 2 (https://github.com/project-chip/connectedhomeip/pull/38283). This implementation Integrate the documentation generation into the CI pipeline `build-and-publish: Build documentation`.

The following changes have been made to support generating ChipDeviceCtrl Python API documentation into de CI pipeline:

- Adds script: `docs/testing/GenerateChipDeviceCtrlDoc.sh`
- Extracts docstrings from: `src/controller/python/chip/ChipDeviceCtrl.py`
- Generates output: `ChipDeviceCtrlAPI.md`
- Updates the GitHub Actions workflow: .`github/workflows/docbuild.yaml`
- Adds generation of the markdown file as a CI step
- Verifies the output file exists using a lightweight bash test

juandediosg, 2025-06-04:
### Current status on Phase 3

The Phase 3 PR Integration of the documentation generation into the CI pipeline has been updated based on feedback and completed (implemented on the PR Phase 2 https://github.com/project-chip/connectedhomeip/pull/38283). This implementation Integrate the documentation generation into the CI pipeline `build-and-publish: Build documentation`.

The following changes have been made to support generating ChipDeviceCtrl Python API documentation into de CI pipeline covering:

**Summary of Implemented Changes/Updates based on feedback:**
- Restored `awk` to avoid MyST cross-reference warnings breaking HTML.
- Added warning suppression in `conf.py` to prevent CI failures.
- Updated CI workflow to remove redundant doc generation.
- Confirmed `.gitignore` properly excludes generated docs files (`docs/_build`).

juandediosg, 2025-06-10:
### Current status on Phase 2 & 3

The PR **Generate Documentation for python API - Docstring Updates (Only ChipDeviceCtrl) + CI Integration** #38283
has been approved by Google.
Awaiting external second approval to be merged.

closed 2025-06-16 by khodya