project-chip/connectedhomeip #37789: TryLockChipStack unit test fails during Darwin CI

feasel0, 2025-02-26:
### Reproduction steps

`PlatformMgr().TryLockChipStack` returns false during the unit test `TestPlatformMgr.TryLockChipStack`in the CI job `Build on Darwin (clang, simulated)`.

```
[ RUN      ] TestPlatformMgr.TryLockChipStack
[1740510125.258] [56655:167053:main] [DL] _Init
src/platform/tests/TestPlatformMgr.cpp:213: Failure
      Expected: locked is true
        Actual: false is true
[1740510125.258] [56655:167053:main] [DL] Inet Layer shutdown
[1740510125.258] [56655:167053:main] [DL] BLE Layer shutdown
[1740510125.258] [56655:167053:main] [DL] System Layer shutdown
[  FAILED  ] TestPlatformMgr.TryLockChipStack
```

For now in [PR #37753](https://github.com/project-chip/connectedhomeip/pull/37753) I'm disabling the check in that unit test when platform is Darwin.

This should be investigated.  If it's determined that it's normal behavior for TryLockChipStack to return false here, and the issue is closed, delete the comment in the gn file mentioning this issue.

### Bug prevalence

Whenever CI runs, after the check is added in PR #37753

### GitHub hash of the SDK that was being used

58738033c6f6eff8884388fd2b00dd70d27dfe85

### Platform

darwin

bzbarsky-apple, 2025-02-27:
darwin does not support locking the stack at all, so TryLockChipStack failing there is _very_ expected.

feasel0, 2025-02-28:
Thanks for the feedback!  Will update the unit test.

closed 2025-02-28 by feasel0