project-chip/connectedhomeip #37832: [BUG] Installion environment fail

emesamwong, 2025-03-01:
### Reproduction steps

I am tring install the Matter environment for development but fail to exeute the install.sh, here is the error message:

[image]

Here is the full logs:
[esp-matter.error.log.txt](https://github.com/user-attachments/files/19036713/esp-matter.error.log.txt)
[clone.esp-matter.log.txt](https://github.com/user-attachments/files/19036716/clone.esp-matter.log.txt)
[check.hash.log.txt](https://github.com/user-attachments/files/19036717/check.hash.log.txt)

I already created an issue in eps-matter: https://github.com/espressif/esp-matter/issues/1302

OS: Ubuntu 22.04.2 LTS
ESP-Matter Commit Id: release/v1.1 (#548ef58)
Connectedhomeip commit Id: #83f4163

### Bug prevalence

always

### GitHub hash of the SDK that was being used

83f4163

### Platform

esp32

andy31415, 2025-03-04:
Looks like a pip error, although I am unclear from what part from the logs.

andy31415, 2025-03-04:
Could you check the contents of `/home/samwong/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/pip-requirements.log` ? Are there more messages there? Otherwise tyring the pip install manually in a venv maybe shows what fails.

My guess is some conflict on versions between modules.

emesamwong, 2025-03-05:
[install fail2_pip-requirements.log](https://github.com/user-attachments/files/19079667/install.fail2_pip-requirements.log)

The first one is the original log, the second one is that I tried to remove the /.environment and run ./install.sh again

emesamwong, 2025-03-05:
[install fail _pip-requirements_part1.txt](https://github.com/user-attachments/files/19079716/install.fail._pip-requirements_part1.txt)
[install fail _pip-requirements_part2.txt](https://github.com/user-attachments/files/19079717/install.fail._pip-requirements_part2.txt)
The original log is too large and fail to upload

andy31415, 2025-03-06:
I see errors compiling gevent.

From what I remember, we used to have issues with libev builds and esp (which requires gdbui which depends on gevent).
https://github.com/project-chip/connectedhomeip/blob/master/integrations/cloudbuild/smoke-test.yaml#L33 for example adds some workarounds, specifically commenting out `gdbgui` from the espressif requirements. you can try to do the same. That code does:

```
perl -i -pe 's/^gdbgui==/# gdbgui==/' /opt/espressif/esp-idf/requirements.txt
```

The path where you have your requirements likely varies.

andy31415, 2025-03-06:
/home/samwong/esp/esp-matter/connectedhomeip/connectedhomeip/scripts/setup/requirements.esp32.txt seems to also be mentioned there having gdbgui.

andy31415, 2025-03-06:
Looking at https://github.com/project-chip/connectedhomeip/blob/master/scripts/setup/requirements.esp32.txt we do not have gdbgui there currently.

andy31415, 2025-03-06:
@emesamwong I believe latest SDK does not use gdbgui so it should not have this issue, however we are also not sufficiently staffed to support such old branches of the SDK (i.e. a backport fix is unlikely to happen from general maintainers side, however maybe espressif as part of their bug report could propose changes).

closed 2025-03-06 by andy31415