project-chip/connectedhomeip #37867: [BUG] IdleModeDuration attribute read failure

agatah2333, 2025-03-04:
### Reproduction steps

I modified the zap file for the virtual device so it could enable the ICDManagement cluster in endpoint 1. However, when I try to read using the command:

```
./chip-tool any read-by-id 0x0046 0x0000 0x654324 1
```
I receive the following error:
```
[1741099120.761] [891979:891979] [TOO] Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)
```

I checked the specification, and the IdleModeDuration (id = 0x0000) should have a default value of 1, so it should return 1 rather than a failure. 

The virtual device returns the log as follows:
```
[1741099120.757] [889126:889126] [IM] Received Read request
[1741099120.757] [889126:889126] [DMG] ReadRequestMessage =
[1741099120.757] [889126:889126] [DMG] {
[1741099120.757] [889126:889126] [DMG]  AttributePathIBs =
[1741099120.757] [889126:889126] [DMG]  [
[1741099120.757] [889126:889126] [DMG]          AttributePathIB =
[1741099120.757] [889126:889126] [DMG]          {
[1741099120.757] [889126:889126] [DMG]                  Endpoint = 0x1,
[1741099120.757] [889126:889126] [DMG]                  Cluster = 0x46,
[1741099120.757] [889126:889126] [DMG]                  Attribute = 0x0000_0009,
[1741099120.757] [889126:889126] [DMG]          }
[1741099120.757] [889126:889126] [DMG] 
[1741099120.757] [889126:889126] [DMG]  ],
[1741099120.757] [889126:889126] [DMG] 
[1741099120.757] [889126:889126] [DMG]  isFabricFiltered = true, 
[1741099120.757] [889126:889126] [DMG]  InteractionModelRevision = 12
[1741099120.757] [889126:889126] [DMG] },
[1741099120.757] [889126:889126] [DMG] IM RH moving to [CanStartReporting]
[1741099120.757] [889126:889126] [DMG] Building Reports for ReadHandler with LastReportGeneration = 0x0000000000000000 DirtyGeneration = 0x0000000000000000
[1741099120.757] [889126:889126] [DMG] <RE:Run> Cluster 46, Attribute 9 is dirty
[1741099120.757] [889126:889126] [DMG] AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_0046 e=1 p=v r=r
[1741099120.757] [889126:889126] [DMG] AccessControl: allowed
[1741099120.757] [889126:889126] [DMG] Reading attribute: Cluster=0x0000_0046 Endpoint=0x1 AttributeId=0x0000_0009 (expanded=0)
[1741099120.757] [889126:889126] [DMG] Failed to read attribute: src/data-model-providers/codegen/CodegenDataModelProvider_Read.cpp:136: IM Error 0x00000501: General error: 0x01 (FAILURE)
[1741099120.757] [889126:889126] [DMG] Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0046, attributeId: 0x0000_0009err = src/data-model-providers/codegen/CodegenDataModelProvider_Read.cpp:136: IM Error 0x00000501: General error: 0x01 (FAILURE)

```

### Bug prevalence

each time 

### GitHub hash of the SDK that was being used

ab3d5aeee87542c4c0e74cdabb34707c44cfc023

### Platform

core, other

bzbarsky-apple, 2025-03-04:
> I modified the zap file for the virtual device so it could enable the ICDManagement cluster in endpoint 1

A few things:

1. No idea what "virtual device" is.
2. ICD management always goes on endpoint 0.

> the IdleModeDuration (id = 0x0000) should have a default value of 1,

That's the value a _client_ should assume if the server does not provide a value.

closed 2025-03-04 by bzbarsky-apple

agatah2333, 2025-03-04:
> > I modified the zap file for the virtual device so it could enable the ICDManagement cluster in endpoint 1
> 
> A few things:
> 
> 1. No idea what "virtual device" is.
> 2. ICD management always goes on endpoint 0 per spec, and the code will register it on endpoint 0 only.  So it sounds like you created a ZAP config that does not match the code, and the ZAP config claims the attribute exists but it does not in fact exist.
> 
> > the IdleModeDuration (id = 0x0000) should have a default value of 1,
> 
> That's the value a _client_ should assume if the server does not provide a value.

The virtual device is chip-all-clusters-app based on Linux. Following your guidance, I enabled the ICD Management on endpoint 0; however, I still cannot read the IdleModeDuration attribute.

bzbarsky-apple, 2025-03-05:
> The virtual device is chip-all-clusters-app based on Linux

OK.

> I enabled the ICD Management on endpoint 0; however, the chip-tool still cannot read the IdleModeDuration attribute.

Did you also enable the relevant compile flags to actually build ICD support, etc?

> Should I manually set the IdleModeDuration as 1 (even 1 is the default value per spec?) when using ZAP 

It doesn't matter what you set in ZAP for this attribute, and in fact it should not allow you to set a default value for it it all, since it's always external, right?