project-chip/connectedhomeip #37880: [BUG] No-Memmorry-Error

emrullahgu, 2025-03-05:
### Reproduction steps

I was able to pair the device while obtaining the PAA certificate from development, but when I tried from production, I got a 'no memory' error. After that, I kept getting the 'no memory' error even when trying from both development and production. I have attached the log.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

4e7ce337fd5fa169c0d06a3792ebb9d8e8274180

### Platform

other

### Platform Version(s)

linux ubuntu 22.04

### Anything else?

[no_memory_error.txt](https://github.com/user-attachments/files/19083506/no_memory_error.txt)

bzbarsky-apple, 2025-03-07:
Device is responding with TableFull, which means it's already commissioned into the maximum number of fabrics it supports.

closed 2025-03-07 by bzbarsky-apple

emrullahgu, 2025-03-07:
> Device is responding with TableFull, which means it's already commissioned into the maximum number of fabrics it supports.
hi  @bzbarsky-apple ,
 
How can I do a factory reset? Because when I try to add my Philips 9290036010 device to Matter through the integrations section in the Wiz app, or when I cycle through the app 3-4 times to get the light to turn blue or purple, I keep encountering a 'no memory' error.

bzbarsky-apple, 2025-03-08:
> How can I do a factory reset?

Depends on your device, no?  See the manual for it?