project-chip/connectedhomeip #37889: [BUG] ICD management cluster attributes 0x0000 0x0009 read failure

agatah2333, 2025-03-05:
### Reproduction steps

Reproduce step:
1. Enable the ICD management cluster in endpoint 0 on chip-all-clusters-app based on Linux.
2. Enable the following variables: chip_enable_icd_lit, chip_icd_report_on_active_mode, and chip_enable_icd_dsls as true in icd.gni , which automatically sets chip_enable_icd_checkin as true in icd/server/build.gn.
3. Compile with ASAN enabled.
4. After pairing with chip-all-clusters-app (node ID: 0x654324), use chip-tool to send the following messages:

```
./chip-tool any read-by-id 0x0046 0x0009 0x654324 0
./chip-tool any read-by-id 0x0046 0x0000 0x654324 0
```
Error Received:
```
[1741208477.262] [3511766:3511766] [TOO] Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)
```
The logs of chip-all-clusters-app and chip-tool are as attached

[chip-all-clusters-app.log](https://github.com/user-attachments/files/19096790/chip-all-clusters-app.log)

[chip-tool.txt](https://github.com/user-attachments/files/19096811/chip-tool.txt)

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

ab3d5ae

### Platform

core

Chapoly1305, 2025-03-05:
I also tried this, seems just moving lit-icd-app stuff to all-cluster-app won't work. Something caused the "Failure" without giving meaningful debug info.

bzbarsky-apple, 2025-03-07:
@mkardous-silabs

bzbarsky-apple, 2025-03-07:
From the all-clusters log:

```
[1741209105.282] [3525750:3525750] [DMG] Failed to read attribute: src/data-model-providers/codegen/CodegenDataModelProvider_Read.cpp:136: IM Error 0x00000501: General error: 0x01 (FAILURE)
```

That line number is what happens when trying to read from the ember data store, which means the AttributeAccessInterface for the ICD cluster was not registered.

@agatah2333 In your all-clusters-app configuration, is `MatterIcdManagementPluginServerInitCallback` being called at startup?

agatah2333, 2025-03-07:
@bzbarsky-apple I have no idea about what the configuration is. I added logs in MatterIcdManagementPluginServerInitCallback(), but it is never being called. It's weird because the lit-icd-app never calls the MatterIcdManagementPluginServerInitCallback().

mkardous-silabs, 2025-03-07:
I think the issue here is the missing `chip_enable_icd_server` build argument. This the main build argument that enables the feature-set.

agatah2333, 2025-03-07:
> [@agatah2333](https://github.com/agatah2333) I think the issue here is the missing `chip_enable_icd_server` build argument. This the main build argument that enables the feature-set.

@mkardous-silabs I already marked the `chip_enable_icd_server` true in the icd.gni file.

mkardous-silabs, 2025-03-07:
> [@mkardous-silabs](https://github.com/mkardous-silabs) I already marked the `chip_enable_icd_server` true in the icd.gni file.

Just making sure since it was not mentionned in the ticket description. Did you get a chance to check if the mandatory attributes work on the all-clusters-app?

Can you validate if the linux lit-icd-app exhibits the same behavior? 
As a side comment, we didn't add support of the ICD in the all-clusters-app deliberately in favor of the lit-icd app

agatah2333, 2025-03-07:
@mkardous-silabs Thanks for your reminder! I edited the description to indicate that chip_enable_icd_server is set to true. I also modified the main.cpp code in chip-all-clusters-app to mimic the lit-icd-app main.cpp code. 

> Can you validate if the linux lit-icd-app exhibits the same behavior?
As mentioned by @Chapoly1305, the lit-icd-app does not experience this problem. Could you please let me know what additional support I should add for using ICD in the all-clusters-app?

bzbarsky-apple, 2025-03-08:
> It's weird because the lit-icd-app never calls the MatterIcdManagementPluginServerInitCallback().

I am sure it does, from generated code....

bzbarsky-apple, 2025-03-08:
@agatah2333 Just to check: did you rerun codegen after modifying the .zap file?

agatah2333, 2025-03-08:
> I am sure it does, from generated code....

Yes, I saw it only shows on PluginApplicationCallbacks.h in lit-icd-app. I even modified the main.cpp code in chip-all-clusters-app to mimic the lit-icd-app main.cpp code. But never shows on chip-all-clusters-app.

> Just to check: did you rerun codegen after modifying the .zap file?

Yes. After modifying the .zap file, I delete the whole out file and rerun the
```
./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-x64-all-clusters-asan-coverage build"
```

agatah2333, 2025-03-08:
> Is there a change of the .matter file associated to the all cluster zap file?
 I also use 
```
$ scripts/tools/zap_regen_all.py
```
When I do this, the .matter file is changed and becomes associated with the all-cluster zap file.

bzbarsky-apple, 2025-03-11:
@agatah2333 The right order of operations is:

1. Modify .zap file
2. Rerun codegen (zap_regen_all works, but you could `./scripts/tools/zap/generate.py examples/all-clusters-app/all-clusters-common/all-clusters-app.zap`)
3. Build, as you had above using `build_examples.py`

agatah2333, 2025-03-11:
@bzbarsky-apple Thanks! Now it works! In step 2, I previously used:
```
$ scripts/tools/zap_regen_all.py
```
but it didn’t work.

Following your instructions, I used:
```
./scripts/tools/zap/generate.py examples/all-clusters-app/all-clusters-common/all-clusters-app.zap
```
and this one worked!

closed 2025-03-12 by agatah2333