project-chip/connectedhomeip #37890: [BUG] Pairing over thread failing on nordic board CHIP Error 0x00000050: Unsupported certificate format

cjandhyala, 2025-03-06:
### Reproduction steps

With SDK ToT (cf45d5d38870a8994ca740b6e3576f5ccc6f9ad8) The pairing script on CI stress rack is failing with CHIP Error 0x00000050: Unsupported certificate format.. . PFA logs.

pairing over thread with chip-tool is failing on nordic board  for the same reason. 

[logs.zip](https://github.com/user-attachments/files/19099962/logs.zip)

[chip_tool_log.txt](https://github.com/user-attachments/files/19099972/chip_tool_log.txt)

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

cf45d5d38870a8994ca740b6e3576f5ccc6f9ad8

### Platform

nrf

Damian-Nordic, 2025-03-06:
@cjandhyala So how is it possible to reproduce it? I'm not sure what CI stress rack is but how many times do you run the commissioning in order to get this error?

Also, the controller logs seem to have truncated lines e.g.
```
2025-03-05 16:42:12,609 - chip.native.ATM - INFO - __init__.py:_RedirectToPythonLogging:41 - 							"NOCValue" : "hex:1530010101240201370324130218260480228127260580254D3A3706241501261182D6F0B91824070124080130094104C79993B3E6336C5B4FBC513FEB902E7BEE805BFE79CAF089815AFCDF5809E0AE9CCF45950315F01A0E07DB57F21A8D7C212CDBB570765403BE5BAE1811212778370A35
```

Would be great if this could be fixed so that we get exact NOC and ICAC values that failed the validation.

cjandhyala, 2025-03-06:
@Damian-Nordic you can reproduce this issue when paired using chip-tool . I attached the chip-tool log as well in the ticket.

PR #37953 (2025-03-11): [nrfconnect] Increase heap size for all-clusters-app
Increase heap size as commissioning fails due to `CHIP_ERROR_NO_MEMORY` error.

Fixes #37890

#### Testing
Tested commissioning locally with `chip-tool`
closed 2025-03-11 by mergify[bot]