project-chip/connectedhomeip #37924: Use `AttributePersistenceProvider`/`SafeAttributePersistenceProvider` for the storage context for server cluster interfaces

andy31415, 2025-03-07:
This is in reference to https://github.com/project-chip/connectedhomeip/pull/37851#discussion_r1985249193

from #37851

We passed in `PersistentStorageDelegate` as a storage context for ServerClusterContext.

instead we should use an attribute persistence provider, since those are the interfaces that are supposed to be available to clusters to read/write their data.

andy31415, 2025-04-04:
We need additional scoping and options for this issue:  when checking changes in #38261 we found that:
- AttributePersistenceProvider is tightly coupled to Ember (uses ember types and native format)
- SafeAttributePersistenceProvider uses a different format and namespace from the AttributePersistenceProvider

As a result, we need a solution that both allows for ember compatibility for clusters already using ember for before (i.e. the binary data format of AttributePersistenceProvider, but without using ember metadata) as well as compatibility with 'AttributeAccessInterface' that is provider by SafeAttributePersistenceProvider.

ratgr, 2025-04-21:
``` mermaid
classDiagram

PersistentStorageDelegate <|-- StorageAdapter
PersistentStorageDelegate <|-- PythonPersistentStorageDelegate
PersistentStorageDelegate <|-- MyServerStorageDelegate
PersistentStorageDelegate <|-- TestPersistentStorageDelegate
PersistentStorageDelegate <|-- KvsPersistentStorageDelegate

CredentialIssuerCommands     *-- PersistentStorageDelegate
EvseTargetsDelegate          *-- PersistentStorageDelegate
FabricAdmin                  *-- PersistentStorageDelegate
IcdManagementAttributeAccess *-- PersistentStorageDelegate
PersistentData               *-- PersistentStorageDelegate

```

ratgr, 2025-04-30:
[image]
Red nodes are those that include Metadata from Ember
Yellow don't really need Ember Metadata but are linked to it
White Nodes don't seem to be a problem or used directly so should be able to Switch from PersistentStorageDelegate to Storage Delegate Wrapper
Green nodes are OK

There is no need to provide an alternative to the normal AttributePersistenceProvider as its only uses seem to be on inside ember implementations

emAfSaveAttributeToStorageIfNeeded
emAfLoadAttributeDefaults

khodya, 2026-11-03:
Implemented as expected

closed 2026-11-03 by khodya