project-chip/connectedhomeip #37939: [Feature] Support for smart desks

mhalano, 2025-03-08:
### Feature description

My idea is to add a device for a smart standing desk. I have one that I use over Smart Life with a Matter Bridge to communicate with Google Home, but the device appears as an air humidifier.
This device has a limited series of options, like up, down and set a memory position, so it's not a complex device, but it shouldn't appear as an air humidifier
Please, let me know if you need any specific technical details about it to implement the feature.

### Platform

all

bzbarsky-apple, 2025-03-14:
That sounds like a spec feature request, not an SDK one....

closed 2025-03-14 by bzbarsky-apple

mhalano, 2025-03-14:
@bzbarsky-apple Where can I create this spec feature request?

bzbarsky-apple, 2025-03-14:
@mhalano Unfortunately, I believe you have to be a Communications Standards Alliance member to do that, or get one of them to do it on your behalf....

bzbarsky-apple, 2025-03-14:
But in general: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/