project-chip/connectedhomeip #38014: [Test Failed] Error when executing networkcommissioning remove-network command in Matter over Thread device

whqxbs, 2025-03-15:
### Test issue(s)

**Problem Background:**
I have added a Matter-over-Thread light to a Thread gateway and paired it with the CHIP Tool. I want to use CHIP Tool to perform some network modification operations on this device. (device Node ID: 1234567 in CHIP Tool)

As shown below, I can use the `networkcommissioning scan-networks` cluster command in the CHIP Tool to scan for Thread Border Routers in the current network.

`chip-tool networkcommissioning scan-networks 1234567 0`
```
[1742023494.777414][53668:53670] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0001
[1742023494.777433][53668:53670] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0001
[1742023494.777476][53668:53670] CHIP:TOO:   ScanNetworksResponse: {
[1742023494.777490][53668:53670] CHIP:TOO:     networkingStatus: 0
[1742023494.777500][53668:53670] CHIP:TOO:     threadScanResults: 2 entries
[1742023494.777514][53668:53670] CHIP:TOO:       [1]: {
[1742023494.777518][53668:53670] CHIP:TOO:         PanId: 38867
[1742023494.777523][53668:53670] CHIP:TOO:         ExtendedPanId: 545067622935023181
[1742023494.777528][53668:53670] CHIP:TOO:         NetworkName: TMSthread_0790
[1742023494.777532][53668:53670] CHIP:TOO:         Channel: 20
[1742023494.777536][53668:53670] CHIP:TOO:         Version: 4
[1742023494.777541][53668:53670] CHIP:TOO:         ExtendedAddress: 5EFD8C295C5CF63C
[1742023494.777546][53668:53670] CHIP:TOO:         Rssi: -67
[1742023494.777549][53668:53670] CHIP:TOO:         Lqi: 255
[1742023494.777554][53668:53670] CHIP:TOO:        }
[1742023494.777562][53668:53670] CHIP:TOO:       [2]: {
[1742023494.777566][53668:53670] CHIP:TOO:         PanId: 53319
[1742023494.777570][53668:53670] CHIP:TOO:         ExtendedPanId: 7450251385094097320
[1742023494.777574][53668:53670] CHIP:TOO:         NetworkName: MyHome1471659479
[1742023494.777578][53668:53670] CHIP:TOO:         Channel: 25
[1742023494.777582][53668:53670] CHIP:TOO:         Version: 4
[1742023494.777586][53668:53670] CHIP:TOO:         ExtendedAddress: CEE30AF6E4910D9F
[1742023494.777590][53668:53670] CHIP:TOO:         Rssi: -72
[1742023494.777594][53668:53670] CHIP:TOO:         Lqi: 255
[1742023494.777598][53668:53670] CHIP:TOO:        }
[1742023494.777602][53668:53670] CHIP:TOO:    }
```
Using the `networkcommissioning read networks` command, it can be observed that the device is currently connected to the Thread Border Router with Network ID `0790781D7B9BCE4D`. When converted to decimal, this Network ID becomes `545067622935023181`, corresponding to the **ExtendedPanId** of the previously mentioned **TMSthread_0790**.

`chip-tool networkcommissioning read networks 1234567 0`
```
[1742023522.912792][53711:53713] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 1627130345
[1742023522.912843][53711:53713] CHIP:TOO:   Networks: 1 entries
[1742023522.912869][53711:53713] CHIP:TOO:     [1]: {
[1742023522.912889][53711:53713] CHIP:TOO:       NetworkID: 0790781D7B9BCE4D
[1742023522.912894][53711:53713] CHIP:TOO:       Connected: TRUE
[1742023522.912900][53711:53713] CHIP:TOO:      }
```

Next, I put the device into arm-fail-safe mode with `chip-tool generalcommissioning arm-fail-safe 30 0 1234567 0` to execute the networkcommissioning remove-network command to remove the existing Thread network.

According to document `11.9.7.6. RemoveNetwork Command - NetworkID Field - This field SHALL contain the NetworkID for the entry to remove: the SSID for Wi-Fi and XPAN ID for Thread.`, I specified the device’s Thread Network ID as the ExtendedPanId using the following commands. However, all three attempts failed (I tried both decimal and hexadecimal formats for the ExtendedPanId in the commands).

`chip-tool networkcommissioning remove-network 545067622935023181 1234567 0`
`chip-tool networkcommissioning remove-network 0790781D7B9BCE4D 1234567 0`
`chip-tool networkcommissioning remove-network 0x0790781D7B9BCE4D 1234567 0`

The `networkingStatus` is error codes 3 as follow. After reviewing `Matter Specification 1.4 11.9.5.4. NetworkCommissioningStatusEnum Type`, it indicates `NetworkIDNotFound - The NetworkID is not among the collection of added networks.`. However, I confirmed that the provided Network ID is correct and exists.

```
[1742023931.126516][54052:54054] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1742023931.126539][54052:54054] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1742023931.126628][54052:54054] CHIP:TOO:   NetworkConfigResponse: {
[1742023931.126669][54052:54054] CHIP:TOO:     networkingStatus: 3
[1742023931.127024][54052:54054] CHIP:TOO:    }
```

I am confused about why the command execution failed. This issue is critical for my Matter over Thread testing. Could the command format be incorrect? I would appreciate a valid command example for successful execution.

Looking forward to your guidance. Thank you!
 

### Platform

all

whqxbs, 2025-03-18:
@Damian-Nordic @bzbarsky-apple @andy31415 
Sorry to bother you all! I really need your help. Thank you very much and look forward to your answers.

Damian-Nordic, 2025-03-18:
Did you try passing the argument as `hex:0790781D7B9BCE4D`?

whqxbs, 2025-03-18:
> Did you try passing the argument as `hex:0790781D7B9BCE4D`?

Oh I forgot about this, it works! Thank you very much for your answer!

closed 2025-03-24 by bzbarsky-apple