project-chip/connectedhomeip #38036: Increase Unit Test Coverage ChipDeviceController

gd-mauri, 2025-03-17:
this ticket covers files:

1. src/controller/ChipDeviceController.h
2. src/controller/ChipDeviceController.cpp

dmaranges-gd, 2025-08-25:
I started working on this ticket last week, but first of all I need to merge the pull request #40453 about "ChipDeviceControllersFactory" files because some methods are tested on these tests.

dmaranges-gd, 2025-09-05:
Currently, I am working on this ticket. I'm examining if I can test all methods using the factory pattern or if I have to perform separate tests.

dmaranges-gd, 2025-09-18:
I added some extra methods to the test plan and are all basic methods tested. To continue on that, I need to reorganize the tests and define a context and the specification of a device. In my opinion, that can take a lot of time. I will try to do that today and make a report to decide if it possible of it is better skip those methods and focus on other files.