project-chip/connectedhomeip #38082: [Feature] How to build the JNI libraries for Linux

jonsmirl, 2025-03-23:
### Feature description

@yunhanw-google  Is there a way to make a Linux build of the kotlin/JNI libraries? That would let me speed up the unit testing of my app code, it is very slow using an Android environment.

### Platform

all

yunhanw-google, 2025-04-01:
./scripts/build/build_examples.py --target linux-x64-java-matter-controller build

this can generate linux x64 jni libs

thanks

closed 2025-04-01 by yunhanw-google

law-ko, 2025-05-21:
```
2025-05-21 15:35:04 INFO    [72/704] c++ obj/third_party/connectedhomeip/src/inet/libInetLayer.InetError.cpp.o
2025-05-21 15:35:04 INFO    ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/Users/lawrenceko/Local/connectedhomeip/./scripts/build/build_examples.py", line 248, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/Users/lawrenceko/Local/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lawrenceko/Local/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Users/lawrenceko/Local/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lawrenceko/Local/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lawrenceko/Local/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lawrenceko/Local/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lawrenceko/Local/connectedhomeip/./scripts/build/build_examples.py", line 238, in cmd_build
    context.obj.Build()
  File "/Users/lawrenceko/Local/connectedhomeip/scripts/build/build/__init__.py", line 75, in Build
    builder.build()
  File "/Users/lawrenceko/Local/connectedhomeip/scripts/build/builders/builder.py", line 107, in build
    self._build()
  File "/Users/lawrenceko/Local/connectedhomeip/scripts/build/builders/gn.py", line 116, in _build
    self._Execute(cmd, title='Building ' + self.identifier)
  File "/Users/lawrenceko/Local/connectedhomeip/scripts/build/builders/builder.py", line 112, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/Users/lawrenceko/Local/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['ninja', '-C', '/Users/lawrenceko/Local/connectedhomeip/out/darwin-arm64-java-matter-controller'] failed: 1

```

@jonsmirl  Not working for me as well, `jni.h` is missing and I am using Android Studio's JAVA_HOME path.