project-chip/connectedhomeip #38094: [Feature] How to get  PAA for real device and how can use it in android example code

sandeep14, 2025-03-24:
### Feature description

I am testing with a real device, and at that time, I am receiving the 'Attestation Information' command from the device: error 101. Please refer to the AttestationVerificationResult enum to understand the errors. I know this is related to the missing PAA. How can I obtain the PAA for my device? I am using a Matter gateway, and in the CHIP tool, we can pass the --paa-trust-store-path in the command line. However, how can we use it in an Android project? Please guide me

### Platform

android

tcarmelveilleux, 2025-03-24:
You cannot "obtain a PAA". A PAA is a trusted root that is maintained by one of the Matter working group participants. You have to engage with one such operator (or become one yourself by contacting the CSA) to issue certificates.

closed 2025-03-24 by tcarmelveilleux

sandeep14, 2025-03-25:
Ok just guide me, how can we use it with example Android project

sandeep14, 2025-03-28:
if I get paa file from vendor or by contacting the CSA then how can use it chiptool android project. @tcarmelveilleux  also there is no documentation to create paa and use it in android example (chiptool) project. Please guide me on that also