project-chip/connectedhomeip #38150: [Build] EmberAFStatus Not Declared in ESP32 Matter Lightning App (ZAP Issues)

kp8620, 2025-03-27:
### Build issue(s)

###Context:
1.I am working with a previous teams Lightning App, they modified it back in early 2024 and told me to take the lightning app download, and just copy their main into it. (I have fixed CMAKE once, aside that and main, everything is up tVeriio date) which is causing issues
2. When building the project, I receive this issue from my AppTask.cpp
3

###Additonal Info:
1. Using Esp-idf V5.2.5
2. Should be the latest Matter-SDK (Downloaded from VsCode)
3. MacOS Ventura

###Steps Taken:
Verified 'basic-types.h' exists
Checked CMAKELists.txt
Regenerated 'zap-generated'  files via ZAP
Verified 'app-common/zap-generated' exists but 'basic-types.h' is not being included correcly/it is but EmberAfStatus is not recognized'

##Additional Context:
I was advised to open this by @shubhamdp 

### Platform

esp32

### Anything else?

Here is the issue: (I Have attached my file as well, I have converted it from cpp to txt) 
..'void AppTask::UpdateClusterState()':
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:198:5: error: 'EmberAfStatus' was not declared in this scope

[AppTask.txt](https://github.com/user-attachments/files/19490184/AppTask.txt)

  198 |     EmberAfStatus status = Clusters::OnOff::Attributes::OnOff::Set(kLightEndpointId, AppLED.IsTurnedOn());
      |     ^~~~~~~~~~~~~
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:198:5: note: maximum limit of 1000 namespaces searched for 'EmberAfStatus'
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:200:9: error: 'status' was not declared in this scope; did you mean 'stat'?
  200 |     if (status != EMBER_ZCL_STATUS_SUCCESS)
      |         ^~~~~~
      |         stat
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:200:9: note: maximum limit of 1000 namespaces searched for 'status'
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:200:19: error: 'EMBER_ZCL_STATUS_SUCCESS' was not declared in this scope
  200 |     if (status != EMBER_ZCL_STATUS_SUCCESS)
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:200:19: note: maximum limit of 1000 namespaces searched for 'EMBER_ZCL_STATUS_SUCCESS'
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:206:5: error: 'status' was not declared in this scope; did you mean 'stat'?
  206 |     status = Clusters::LevelControl::Attributes::CurrentLevel::Set(kLightEndpointId, AppLED.GetLevel());
      |     ^~~~~~
      |     stat
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:206:5: note: maximum limit of 1000 namespaces searched for 'status'
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:208:19: error: 'EMBER_ZCL_STATUS_SUCCESS' was not declared in this scope
  208 |     if (status != EMBER_ZCL_STATUS_SUCCESS)
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/main/AppTask.cpp:208:19: note: maximum limit of 1000 namespaces searched for 'EMBER_ZCL_STATUS_SUCCESS'
[1279/1367] Building CXX object esp-idf/main/CMakeFiles/__idf...connectedhomeip/examples/platform/esp32/ota/OTAHelper.cpp.obj
In file included from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributePathExpandIterator.h:22,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/ReadHandler.h:28,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/ReportScheduler.h:24,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/TimerDelegates.h:20,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/server/Server.h:71,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/ota-requestor/DefaultOTARequestor.h:26,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32/ota/OTAHelper.cpp:21:
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model-provider/Provider.h:101:50: warning: missing initializer for member 'chip::app::DataModel::InteractionModelContext::dataModelChangeListener' [-Wmissing-field-initializers]
  101 |     InteractionModelContext mContext = { nullptr };
      |                                                  ^
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model-provider/Provider.h:101:50: warning: missing initializer for member 'chip::app::DataModel::InteractionModelContext::actionContext' [-Wmissing-field-initializers]
[1280/1367] Building CXX object esp-idf/main/CMakeFiles/__idf...homeip/examples/platform/esp32/common/Esp32ThreadInit.cpp.obj
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32/common/Esp32ThreadInit.cpp: In function 'void ESPOpenThreadInit()':
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32/common/Esp32ThreadInit.cpp:44:5: warning: missing initializer for member 'esp_openthread_radio_config_t::<anonymous>' [-Wmissing-field-initializers]
   44 |     };
      |     ^
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32/common/Esp32ThreadInit.cpp:44:5: warning: missing initializer for member 'esp_openthread_host_connection_config_t::<anonymous>' [-Wmissing-field-initializers]
[1281/1367] Building CXX object esp-idf/main/CMakeFiles/__idf.../examples/platform/esp32/common/CommonDeviceCallbacks.cpp.obj
In file included from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributePathExpandIterator.h:22,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/ReadHandler.h:28,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/ReportScheduler.h:24,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/TimerDelegates.h:20,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/server/Server.h:71,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32/common/CommonDeviceCallbacks.cpp:25:
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model-provider/Provider.h:101:50: warning: missing initializer for member 'chip::app::DataModel::InteractionModelContext::dataModelChangeListener' [-Wmissing-field-initializers]
  101 |     InteractionModelContext mContext = { nullptr };
      |                                                  ^
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model-provider/Provider.h:101:50: warning: missing initializer for member 'chip::app::DataModel::InteractionModelContext::actionContext' [-Wmissing-field-initializers]
[1282/1367] Building CXX object esp-idf/main/CMakeFiles/__idf...dhomeip/examples/platform/esp32/common/Esp32AppServer.cpp.obj
In file included from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributePathExpandIterator.h:22,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/ReadHandler.h:28,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/InteractionModelEngine.h:45,
                 from /Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32/common/Esp32AppServer.cpp:22:
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model-provider/Provider.h:101:50: warning: missing initializer for member 'chip::app::DataModel::InteractionModelContext::dataModelChangeListener' [-Wmissing-field-initializers]
  101 |     InteractionModelContext mContext = { nullptr };
      |                                                  ^
/Users/demeter/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model-provider/Provider.h:101:50: warning: missing initializer for member 'chip::app::DataModel::InteractionModelContext::actionContext' [-Wmissing-field-initializers]
[1284/1367] Building CXX object esp-idf/main/CMakeFiles/__idf...d/app-common/app-common/zap-generated/cluster-objects.cpp.obj
ninja: build stopped: subcommand failed.

bzbarsky-apple, 2025-03-27:
EmberAfStatus was removed in https://github.com/project-chip/connectedhomeip/pull/32055

You should use Protocols::InteractionModel::Status instead.

bzbarsky-apple, 2025-03-27:
And the relevant `Status::Whatever` values instead of `EMBER_ZCL_STATUS_WHATEVER` for the constants.

closed 2025-03-27 by bzbarsky-apple