project-chip/connectedhomeip #38162: [BUG] Early ble-wifi commissioning failure when running bluez 5.79

enkiusz, 2025-03-28:
### Reproduction steps

### Summary

Issue is the inability to commission a device running bluez 5.79-2 shipped in [Debian Trixie](https://www.debian.org/releases/trixie/) using ble-wifi.

The debian trixie image used is https://raspi.debian.net/tested/20231111_raspi_4_trixie.img.xz
`c2301aec3c44c5f3c96334ff81d4d4f0f4d9370e0f7fdf1ed5aa53ae84c10fdd  20231111_raspi_4_trixie.img.xz`

The kernel used is https://github.com/raspberrypi/linux.git branch rpi-6.6.y (HEAD 0553897d77e849a86e836ddf1e0c0dbbd8e64f83) with applied BLE fix:

https://lore.kernel.org/linux-bluetooth/174180783150.904159.8994265489515508641.git-patchwork-notify@kernel.org/T/#t

This kernel has the uname raspberry-pi 6.6.78-e9c96463a234 below.

The issue appears when the 5.79-2 bluez is in the device role. When the bluez binary is replaced with 5.66-1+rpt1+deb12u2 copied from debian 12 the commissioning is successful.

### Devices involved

#### polandball
- Raspberry Pi 4 Model B Rev 1.2
- debian 12 (bookworm)
- kernel 6.6.78-e9c96463a234
- build-in BLE adapter
#### blackberry
- Raspberry Pi 3 Model B Rev 1.2
- debian trixie
- kernel 6.6.78-e9c96463a234
- built-in BLE adapter

### Test scenario matrix

| Controller →→→ Device ↓↓↓              | polandball / bluez 5.66-1+rpt1+deb12u2 | blackberry / bluez 5.79-2 | blackberry / bluez 5.66-1+rpt1+deb12u2 |
| -------------------------------------- | -------------------------------------- | ------------------------- | -------------------------------------- |
| polandball / bluez 5.66-1+rpt1+deb12u2 |                                        | Good logs 1               |                                        |
| blackberry / bluez 5.79-2              | failure mode 1                         |                           |                                        |
| blackberry / bluez 5.66-1+rpt1+deb12u2 | OK (no logs)                           |                           |                                        |

### Reproduction commands (only for reference, internal tooling)

```
blackberry:
rm *.log; rm -f /tmp/chip*; out-ubuntu-24.04/linux-arm64-light-clang/chip-lighting-app --wifi 2>&1 | tai64n | pv > rpi3-debian-trixie-lighting-app.log
rm *.log; /usr/libexec/bluetooth/bluetoothd --nodetach --experimental 2>&1 | tai64n | pv > rpi3-debian-trixie-lighting-app-bluez.log
 
polandball:
rm *.log; rm /tmp/chip*; out-ubuntu-22.04/linux-arm64-chip-tool-clang/chip-tool pairing ble-wifi 1 XXXXX XXXXX 20202021 3840 2>&1 | tai64n | tee rpi4-debian-12-chip-tool.log
rm *.log; /usr/libexec/bluetooth/bluetoothd --nodetach --experimental --debug 2>&1 | tai64n | pv > rpi4-debian-12-chip-tool-bluez.log
 
parse & merge:
 
chip-logs-format -P chip --no-time --no-pid --log-name lighting-app:rpi3-debian-trixie-lighting-app.log
tai64n-merge --no-tai64 rpi3-debian-trixie-lighting-app.log rpi3-debian-trixie-lighting-app-bluez.log > rpi3-debian-trixie.log
```

### Logs

#### Good logs 1

[good.tar.gz](https://github.com/user-attachments/files/19504272/good.tar.gz)

#### Failure mode 1

[failure mode 1.tar.gz](https://github.com/user-attachments/files/19504274/failure.mode.1.tar.gz)

### Observation

The good log contains evidence that bluez received the indication confirmation messages from HCI:
```
bluetoothd[874971]: src/gatt-database.c:acquire_notify_reply() AcquireNotify success: fd 31 MTU 517
bluetoothd[874971]: src/gatt-database.c:send_notification_to_device() GATT server sending indication
bluetoothd[874971]: src/gatt-database.c:conf_cb() GATT server received confirmation
lighting-app I:-:BLE: Indication confirmation, 0x7f9c034410
lighting-app I:-:BLE: entered HandleGattSendConfirmationReceived
lighting-app I:-:BLE: entered HandleHandshakeConfirmationReceived
lighting-app I:-:BLE: got peripheral handshake indication confirmation
lighting-app I:-:BLE: entered DriveSending
lighting-app I:-:BLE: exiting HandleHandshakeConfirmationReceived
bluetoothd[874971]: src/shared/att.c:can_read_data() (chan 0x557c9065b0) ATT PDU received: 0x12
bluetoothd[874971]: src/shared/gatt-server.c:write_cb() Write Req - handle: 0x0042
bluetoothd[874971]: src/shared/gatt-server.c:write_complete_cb() Write Complete: err 0
```

This seems to be missing from failure mode 1 logs causing the next steps of communication to not send next messages:
```
lighting-app I:-:BLE: entered DriveSending
lighting-app I:-:BLE: NO SEND: Gatt op in flight
lighting-app I:-:BLE: starting send-ack timer
lighting-app I:-:BLE: entered StartSendAckTimer
lighting-app I:-:BLE: starting new SendAckTimer
lighting-app I:-:BLE: reassembled whole msg, len = 98
lighting-app I:-:EM: >>> [E:24805r S:0 M:240612342] (U) Msg RX from 0:97D93795B433734A [0000] to 0000000000000000 --- Type 0000:20 (SecureChannel:PBKDFParamRequest) (B:98)
lighting-app I:-:EM: Handling via exchange: 24805r, Delegate: 0x55934330a0
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

7d0479661d

### Platform

raspi

enkiusz, 2025-03-28:
This is a duplicate of https://github.com/project-chip/connectedhomeip/issues/37359

bzbarsky-apple, 2025-04-08:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/37359

closed 2025-04-08 by bzbarsky-apple