project-chip/connectedhomeip #38193: [BUG] OffWithEffect implementation is inconsistent and out of sync with the behavious of other lighting control cluster.

andrew-lifx, 2025-04-01:
### Reproduction steps

For any light implementation honouring the lighting cluster attributes, when the _OffWithEffect_ commands of the On/Off Cluster are supported by simply providing the `OnOffEffect` object and a handler to process the effect request, the effect do not work.
Instead the light simply turns off.

### Bug prevalence

Whenever any OffWithEffect Command is used.

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

esp32

### Anything else?

For most of the lighting Clusters (On/Off, Level Control, Colour Control) the SDK implementation its own handling of transition timing and attribute updates. Attribute values are updated  frequently enabling the device light control to render whatever is required by the Matter control. For the OffWithEffect implementation this has apparently been handed off to the application code (in a similar way to Identify effects).
There is no synchronisation with application on the progress of the effect. After calling the effect trigger handler it then immediate sets the State to Off. For any application relying on the attribute callbacks provide Matter control of the light, this immediate turns the light off and no effect will be observed.
This can be seen in the code reference below.
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/on-off-server/on-off-server.cpp#L677-L691

Managing this lack of synchronisation from Matter can be handled by holding off or ignoring the State change, but this behaviour is inconsistent with the processing of other commands by the On/Off, Level Control and Colour Control clusters. It also means that Matter On/Off State temporarily does not represent the actual light state, but rather the state the light is transitioning to. Again inconsistent with the other commands.
There is also nothing to stop the SDK receiving other commands during the Off Effect, and will consider the light to be Off when it it actually On. And what level does Matter expect the light to be at? Does the application code need to update the CurrentLevel attribute while implementing the effect?
It is unclear what is expected of the application (client) in handling these Effects.

As a side note, there is a similar issue with implementation of Identify effects. However there is normally no obvious failure since the SDK does not change the On/Off State or any of the other cluster attributes, leaving the application to do what needs to be done.

bzbarsky-apple, 2025-04-08:
@jmartinez-silabs

jmartinez-silabs, 2025-04-09:
@andrew-lifx While I can understand your point. The OnOff cluster spec doesn't mention any direct interaction with the LevelControl CurrentLevel attribute for the OffWithEffect from what I can tell. It also doesn't mention any delay with setting the OnOff attribute to False upon receiving the OffWithEffect command.

> 7.4.3. Effect on Receipt
> On receipt of the OffWithEffect command the server SHALL check the value of the GlobalSceneControl attribute.
> 
> If the GlobalSceneControl attribute is equal to TRUE, the server SHALL store its settings in its global scene then set the GlobalSceneControl attribute to FALSE, then set the OnOff attribute to FALSE and if the OnTime attribute is supported set the OnTime attribute to 0.
> 
> If the GlobalSceneControl attribute is equal to FALSE, the server SHALL only set the OnOff attribute to FALSE.

The spec does describe what the define effect must be but only mentions a `SHOULD` for the actual trigger act.

> This field is used to indicate which variant of the effect, indicated in the EffectIdentifier field, SHOULD be triggered. If the server does not support the given variant, it SHALL use the default variant. This field is dependent on the value of the EffectIdentifier field and SHALL contain one of the non-reserved values listed in either [DelayedAllOffEffectVariantEnum](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/OnOff.adoc#ref_DelayedAllOffEffectVariantEnum) or [DyingLightEffectVariantEnum](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/OnOff.adoc#ref_DyingLightEffectVariantEnum).

To my understanding. The effect application is the responsibility of the application/manufacturer code through `OnOffEffect .mOffWithEffectTrigger` and should not directly affect the LevelControll CurrentLevel. 

The behaviour you are describe/desire would require spec chang first.

andrew-lifx, 2025-04-10:
@jmartinez-silabs You make a fair point. And I did considered raising an issue with the spec, but the problem is not so much with what is does say, but what it doesn't say. In the absence of clarity it is difficult to understand the intent of the Spec sometimes.
It is easer to start with the SDK and determine if the behaviour is as intended, and if not, is that because it did not follow the intent of the Spec or because of an error in the implementation or an error in the spec.

In this case the SDK may be a valid implementation of the current Spec but that doesn't mean it is a good implementation or easy to work with.
Your comment about the **SHOULD** I would argue is in reference to the actual effect not the triggering of the effect, especially when taken with ".. it SHALL use the default variant.". The "Effect on Receipt" section does not actually say anything about _triggering_ the effect.
If we were being pedantic it could be argued there is no requirement to actually perform the effect, just to support the command. But this is just plain silly.

The "Effect on Receipt" section makes no mention of when the OnOff attribute should be set to False, just that it SHALL be set. It is not explicitly clear in the Spec, but it would seem to me the implication of the effect descriptions is that there is a delayed Off and therefore the setting of the OnOff attribute would also be delayed. The delay being till the end of the effect. This behaviour would be consistent with the standard Off transition.
> 1.5.7.1. Off Command
1.5.7.1.1. Effect on Receipt
On receipt of the Off command, a server SHALL set the OnOff attribute to FALSE.
Additionally, when the OnTime attribute is supported, the server SHALL set the OnTime attribute to 0.

There is no explicit reference to a delay in the setting of the On/Off attribute for the Off Command, however this is how the SDK is implemented. It would would say is the intent.

As for interaction with the LevelControl CurrentLevel attribute a large hole the Matter Spec is its reflection of the state of the physical device. Instead it has it's own virtual representation using the cluster attributes. Rarely does it link this to the physical device. Nowhere have I seen any consideration of any external (to Matter) control of the device. Matter seems to live in a world of it's own, mostly oblivious to the physical device. In all our devices Matter is not the only control interface for the light. The light may change state, brightness or colour from other interfaces. To keep Matter's internal representation in sync we need to constantly update it with the actual state of the device. This is our issue, but Matter SDK does not make it easy for us, especially when we have inconsistencies in the interface behaviour.
In our case any change of the Light irrespective of where it comes from will be reflected back into the Matter State. This includes any effects and transitions.
Whether this is what is expected or intended is not clear.

We could argue about what the Spec should or shouldn't say for a long time. Irrespective of this the OffWithEffect implementation is inconsistent with other parts of the SDK, and difficult to interface with as a result of this inconsistency. Its behaviour also invalidates the conventions for the interface to other cluster attributes. We can no longer take the value changes as an indication of the changes to the physical light.

I believe it is possible to fix these inconsistencies without changing the Spec. 
However, Spec change would help to clarify the intent.

jmartinez-silabs, 2025-04-10:
@andrew-lifx 
> 1.5.7.1. Off Command
> 1.5.7.1.1. Effect on Receipt
> On receipt of the Off command, a server SHALL set the OnOff attribute to FALSE.
> Additionally, when the OnTime attribute is supported, the server [...]

The Level control spec, however, has a clear section about the effect on On Off Toggle command. This is all linked through the LT featuremap

https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/LevelControl.adoc#411-effect-of-onoff-commands-on-the-currentlevel-attribute

There is still no mention of the OffEffectWithEffect.

While I understand and agree with your overall goal and concept, I disagree with your approach to implementing these changes in SDK first. The SDK should always be implemented exactly as the Spec mandates. If the Spec is unclear about something, the SDK should not make arbitrary assumptions about its behavior.

Two individuals interpreting the same ambiguous text can arrive at different conclusions. For instance, if we modify the SDK to behave as you suggest the OffEffectWithEffect should, based on your input, another person might later raise a ticket arguing that the Spec doesn't specify X and SDK should instead do Y. This would lead to endless churn.

In my view, the current implementation of OffWithEffect, although perhaps not ideal, adheres to what the Spec dictates. Therefore, any changes to the SDK should be preceded by changes to the Spec.

I agree that the Spec should be more detailed and clearer about the flexibility allowed to app makers and the strictness required of the SDK with all the onoff / level control interactions.

andrew-lifx, 2025-04-11:
Sorry, I am not suggesting making the changes in the SDK first, unless there is something the SDK has clearly not implemented correctly. But examining the SDK implementation and how it does, could or should work will given better perspective on what the issues with the spec are.
I don't fully understand the intent of the spec, therefore it is difficult for me to say what should change. There are a number of questions about how to interpret the spec I do not have answers to.
But I can say what is problematic with the SDK implementation, where this is the fault of the implementation or the Spec I don't know.
The Spec will never be perfect no mater how much detail it has.
I do not believe simply adhering to the Spec is sufficient. You can adhere to the Spec and have some that is completely non-functional. The implementation needs to Adhere to the Spec AND be functional.
There aught to be consistent interpretation convention, written or otherwise, that the implementation follows. If there is conventions being applied here then we should discuss what they are and whether they are consistent.
At the moment we have to hack the SDK to make this feature work, or leave it as-as and say we tried but it is broken in Matter.
There are always many different ways of implementing a feature, but the behaviour should always be same.

Getting back to this specific issue, questions are:
Is this the correct behaviour of the OffWithEffect and what is the basis for the conclusion?
Is this how it was intended to behave? If not then we should consider it broken and look for how this might be resolved.
If this is how it is intended to behave, then I need some clarification on how our device is expected to deal with seemingly inconsistent information that we are receiving from the Matter SDK. (Specifically the effect trigger and On/Off indications, and potentially any other state changes that may come while performing an effect.)

jmartinez-silabs, 2025-04-11:
@andrew-lifx Why can you not use the `mOffWithEffectTrigger` callback to execute the effect on your Physical light. If the OnOff state returns to ON you cancel the Triggerred effect. Why does that require SDK changes?
Sure, it might requires a few added checks in your app code but it doesn't lead to SDK changes

andrew-lifx, 2025-04-14:
We can and currently do something like this, but it is unnecessarily complicated, because while the effect is running, Matter is  out or sync with the rest our light control and we have to manage this somehow.
There are many different ways we can implement a work-around for this. The problem is that they either add complexity and/or risk of other issues.

In my opinion the implementation of the OffWithEffect support in the SDK is inconsistent with the implementation of other clusters and commands, and makes the task of integrating the SDK more difficult for manufacturers than it needs to be.
With the constraints on resources we have in an embedded environment, the simplest and least risky solution, from my perspective, would be to change the SDK implementation to be more consistent with the behaviour of the other clusters and commands.

You may disagree, or have reasons for a different approach. That's fine. I just need to understand if the issue I see is going to be accepted and addressed in some way, or what the reason is for it behaving this way.
Based on that I can determined the best way for us to work around the issue. We already have a number of patches to the SDK, that we maintain. Some are backports of fixes. Some are platform changes. Some are simple unresolved bugs.