project-chip/connectedhomeip #38239: Occasionally, when re - initializing the ChipDeviceController, an error message "No memory" appears, which causes the app to crash unexpectedly

CaoJieNan, 2025-04-03:
### Reproduction steps

My app has two Fabrics. Occasionally, when I switch from the first Fabric to the second one, an error message "No memory" appears, causing the app to crash unexpectedly. The error log shows: "FabricTable.cpp:833: CHIP Error 0x0000000B: No memory".

Steps to reproduce:
1. Launch the app, initialize the first Fabric, and create a ChipDeviceController.
2. Call the shutdownCommissioning method to destroy the first Fabric.
3. Initialize the second Fabric and create a ChipDeviceController.
4. The app crash issue occurs.

Development environment:
Android 12

Error Logs: 
> 04-02 10:50:21.804 I/CTL     (18271): newDeviceController() called
> 04-02 10:50:21.804 I/CTL     (18271): Setting attestation nonce to random value
> 04-02 10:50:21.804 I/CTL     (18271): Setting CSR nonce to random value
> 04-02 10:50:21.804 I/CTL     (18271): Setting attestation nonce to random value
> 04-02 10:50:21.804 I/CTL     (18271): Setting CSR nonce to random value
> 04-02 10:50:21.804 I/CTL     (18271): KVS: Getting key g/icdfl
> 04-02 10:50:21.804 I/CTL     (18271): Setting attestation nonce to random value
> 04-02 10:50:21.804 I/CTL     (18271): Setting CSR nonce to random value
> 04-02 10:50:21.804 I/CTL     (18271): KVS: Getting key AndroidDeviceControllerKey
> 04-02 10:50:21.805 D/BaseKeypairDelegate(18271): ======================== getPublicKey
> 04-02 10:50:21.805 E/CTL     (18271): Device Controller Factory already initialized...
> 04-02 10:50:21.805 I/CTL     (18271): KVS: Getting key f/11/r
> 04-02 10:50:21.805 D/KeyValueStoreManager(18271): Key 'f/11/r' not found in shared preferences
> 04-02 10:50:21.805 I/CTL     (18271): KVS: Getting key f/11/n
> 04-02 10:50:21.805 D/KeyValueStoreManager(18271): Key 'f/11/n' not found in shared preferences
> 04-02 10:50:21.805 I/CTL     (18271): KVS: Getting key f/11/i
> 04-02 10:50:21.805 D/KeyValueStoreManager(18271): Key 'f/11/i' not found in shared preferences
> 04-02 10:50:21.805 E/FP      (18271): Reverting pending fabric data for fabric 0x11
> 04-02 10:50:21.805 E/FP      (18271): Reverting pending fabric data for fabric 0x11
> 04-02 10:50:21.805 I/TS      (18271): Pending Last Known Good Time: 2025-03-26T08:29:48
> 04-02 10:50:21.805 I/CTL     (18271): KVS: Getting key g/lkgt
> 04-02 10:50:21.805 I/TS      (18271): Previous Last Known Good Time: 2025-03-26T08:29:48
> 04-02 10:50:21.805 I/TS      (18271): Reverted Last Known Good Time to previous value
> 04-02 10:50:21.806 W/System.err(18271): chip.devicecontroller.ChipDeviceControllerException: src/credentials/FabricTable.cpp:833: CHIP Error 0x0000000B: No memory
> 04-02 10:50:21.806 W/System.err(18271): 	at chip.devicecontroller.ChipDeviceController.newDeviceController(Native Method)
> 04-02 10:50:21.806 W/System.err(18271): 	at chip.devicecontroller.ChipDeviceController.<init>(ChipDeviceController.java:69)
> 04-02 10:50:21.806 W/System.err(18271): 	at com.rainmaker.ChipClient.getChipDeviceController(ChipClient.kt:51)
> 04-02 10:50:21.806 W/System.err(18271): 	at com.xxx.createMatterClient(xxxPlugin.kt:383)
> 04-02 10:50:21.806 W/System.err(18271): 	at com.xxx.onMethodCall(xxxPlugin.kt:128)
> 04-02 10:50:21.806 W/System.err(18271): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:267)
> 04-02 10:50:21.806 W/System.err(18271): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:292)
> 04-02 10:50:21.806 W/System.err(18271): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
> 04-02 10:50:21.806 W/System.err(18271): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
> 04-02 10:50:21.806 W/System.err(18271): 	at android.os.Handler.handleCallback(Handler.java:938)
> 04-02 10:50:21.806 W/System.err(18271): 	at android.os.Handler.dispatchMessage(Handler.java:99)
> 04-02 10:50:21.806 W/System.err(18271): 	at android.os.Looper.loopOnce(Looper.java:233)
> 04-02 10:50:21.806 W/System.err(18271): 	at android.os.Looper.loop(Looper.java:344)
> 04-02 10:50:21.806 W/System.err(18271): 	at android.app.ActivityThread.main(ActivityThread.java:8212)
> 04-02 10:50:21.806 W/System.err(18271): 	at java.lang.reflect.Method.invoke(Native Method)
> 04-02 10:50:21.806 W/System.err(18271): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
> 04-02 10:50:21.806 W/System.err(18271): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1034)

### Bug prevalence

1-2 times a week

### GitHub hash of the SDK that was being used

577b3f2be0e526643c124071de64f2e810982c87

### Platform

android

bzbarsky-apple, 2025-04-08:
> Call the shutdownCommissioning method to destroy the first Fabric.

This shuts down the controller _but does not remove the fabric from the fabric table_.  So it does not in fact "destroy the Fabric".  Eventually the fabric table fills up and no more new fabrics can be added.

Sounds like the relevant APIs need fixing to either remove the fabric or at least add a way to remove it.

CaoJieNan, 2025-04-30:
Hey, is there any commit update right now that can solve this problem?