project-chip/connectedhomeip #38298: [CERT-TEST-FAILURE] TC-OPSTATE-2.5 - Python scripts to b update as per the latest Test Plan

arulkumargrl, 2025-04-08:
### Feature Area

Area:OperationalState

### Test Case

TC-OPSTATE-2.5

### Reproduction steps

We reviewed the Python script against the latest test plan. Based on the current updates, the Python script for TC-OPSTATE-2.5 needs to be updated accordingly.

**Script Reference** 
[image]

**Test Plan Reference**

[image]

Test plan PR https://github.com/CHIP-Specifications/chip-test-plans/pull/4890 
https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_OpstateCommon.py
https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_OPSTATE_2_5.py

### Bug prevalence

--

### GitHub hash of the SDK that was being used

-----

### Platform

raspi

marktrayer, 2025-04-24:
Test Script PR has been merged

KishokG, 2025-04-28:
@marktrayer Still the script is not exactly matching with the test plan. For example in test plan I can see step 2a, 2b, 14a, 14b and 23 which is not available in the script. Could you please check your side and raise the PR?

KishokG, 2025-05-16:
@marktrayer In the below mentioned snapshot you can see the line(Step 5 in TEST_TC_OPSTATE_BASE_2_5 section of TC_OpstateCommon.py) : if initial_countdown_time is not NullValue: is mentioned outside of the if condition but it should be there inside the if condition.

[image]

commit 7fd58854 (2025-06-13): Disable WEBRTCP TCs 2.1 and 2.3 (#39528)

* Cleanup step definitions, and marking of test steps
Fixes #38298

* Restyled by autopep8

* Fix minor typos

* Fix numbering for step skip

* Mandate if the revision is equal to or greater than the baseline

* Clean up bad merge

* Correct erroneous method signature

* Remove WEBRTCP 2.1 and 2.3 from CI

* Remove erroneous file

* Add link to issue

* Remove ":"

* Restyled by whitespace

* Restyled by prettier-yaml

---------
PR #39365 (2025-06-13): [TC OPSTATE 2_5] Address test step issues
Fixes #38298 

#### Testing

Ensure python wheel is current
Run updated OPSTATE 2_5 against all-clusters
commit 40b5fce0 (2025-06-13): [TC OPSTATE 2_5] Address test step issues (#39365)

* Cleanup step definitions, and marking of test steps
Fixes #38298

* Restyled by autopep8

* Fix minor typos

* Fix numbering for step skip

* Mandate if the revision is equal to or greater than the baseline

* Clean up bad merge

* Correct erroneous method signature

---------
closed 2025-06-13 by andy31415