project-chip/connectedhomeip #38315: Python Virtual environment is not building due to addition of wifi PAF libs

Harshith-GRL, 2025-04-09:
Encountering exception when  trying to build python virtual environment on linux amd64 laptop which uses ubuntu os, The command I am using to build the python virtual environment is 
`scripts/build_python.sh -m platform -d true -i test_py_env`

During the virtual environment build process encountering this error from ninja

```
Environment looks good, you are ready to go!

Done. Made 10376 targets from 435 files in 1453ms
ninja: Entering directory `./out/python_lib'
[102/455] c++ obj/src/wifipaf/libWiFiPAFLayer.WiFiPAFEndPoint.cpp.o
FAILED: [code=1] obj/src/wifipaf/libWiFiPAFLayer.WiFiPAFEndPoint.cpp.o 
g++ -MMD -MF obj/src/wifipaf/libWiFiPAFLayer.WiFiPAFEndPoint.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../config/python -I../../zzz_generated/app-common -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../src/wifipaf/WiFiPAFEndPoint.cpp -o obj/src/wifipaf/libWiFiPAFLayer.WiFiPAFEndPoint.cpp.o
../../src/wifipaf/WiFiPAFEndPoint.cpp: In member function 'CHIP_ERROR chip::WiFiPAF::WiFiPAFEndPoint::GetPktSn(chip::Encoding::LittleEndian::Reader&, uint8_t*, chip::WiFiPAF::SequenceNumber_t&)':
../../src/wifipaf/WiFiPAFEndPoint.cpp:813:14: error: conversion from 'size_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
  813 |     SnOffset += kTransferProtocolHeaderFlagsSize;
      |     ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../src/wifipaf/WiFiPAFEndPoint.cpp:816:18: error: conversion from 'size_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
  816 |         SnOffset += kTransferProtocolMgmtOpSize;
      |         ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../src/wifipaf/WiFiPAFEndPoint.cpp:820:18: error: conversion from 'size_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
  820 |         SnOffset += kTransferProtocolAckSize;
      |         ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: error: unrecognized command line option '-Wno-unknown-warning-option' [-Werror]
cc1plus: all warnings being treated as errors
[111/455] c++ obj/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o
ninja: build stopped: subcommand failed.
```
seeing this issue on my sdk sha 4057c17a478f319107f4f20938e63502e6e2ce68

Harshith-GRL, 2025-04-10:
@crlonxp  can you have a look at this ?

crlonxp, 2025-04-10:
The PR is created for this issue: https://github.com/project-chip/connectedhomeip/pull/38334

PR #38334 (2025-04-11): Fix the building error by failing to convert the variable type
#### Testing
$ source scripts/activate.sh
$ ./scripts/build_python.sh -m platform -d true -i test_py_env

Fixes #38315 by changing the variable type from unit8_t to size_t
closed 2025-04-11 by mergify[bot]