project-chip/connectedhomeip #38319: [1.3] createControllerOnNewFabric if no vendorID then failed to create controller

robinmo, 2025-04-09:
### Reproduction steps

if commission a matter device with 11 digit short manual pairing code will fail to createControllerOnNewFabric because of lack of vendorID info.
Then how can we commission with 11 digit short manual pairing code ?

### Bug prevalence

Whenever

### GitHub hash of the SDK that was being used

iOS 18.2

### Platform

darwin

### Platform Version(s)

1.3

### Type

Core SDK Interopability Issue

robinmo, 2025-04-09:
@bzbarsky-apple

bzbarsky-apple, 2025-04-10:
This is the vendor ID _for the controller_, not the vendor ID of some device being commissioned.  The documentation clearly says:

```
/**
 * Vendor ID (allocated by the Connectivity Standards Alliance) for
 * this controller.
 *
 * If not nil, must not be the "standard" vendor id (0).
 *
 * When creating a new fabric:
 *
 * * Must not be nil.
 *
```

So for example, if Apple Home calls createControllerOnNewFabric, it would pass in the Apple Home vendor ID (0x1349).

If the implementor of the controller does not have an assigned vendor ID, it should probably use one of the "test" vendor IDs.

closed 2025-04-10 by bzbarsky-apple