project-chip/connectedhomeip #38320: [TC-TSTAT-2.2] Step 13b and 15b needs to be added in python test script as per the test plan

KishokG, 2025-04-09:
### Reproduction steps

- In TC-TSTAT-2.2, Step 13b and 15b needs to be added in python test script as per the test plan.

- Here is the test plan PR for reference: https://github.com/CHIP-Specifications/chip-test-plans/pull/4630

### Bug prevalence

s

### GitHub hash of the SDK that was being used

s

### Platform

raspi

### Platform Version(s)

s

Pradeep-grl, 2026-02-02:
@KishokG  raised a PR:https://github.com/project-chip/connectedhomeip/pull/38555 to address this issue, and it was subsequently merged. However, the PR was later reverted due to a variable-related issue. 
@nivi-apple / @hasty  Could you please fix this issue?

Ashwinigrl, 2026-09-30:
Script PR:https://github.com/project-chip/connectedhomeip/pull/74482

closed 2026-10-21 by mergify[bot]