project-chip/connectedhomeip #38341: [darwin] Darwin tests with leak detection enabled is flaky

andy31415, 2025-04-10:
It currently seems to fail very frequently (10-50% of the time).

Example errors (grep only, the logs are 450-500MB):

https://github.com/project-chip/connectedhomeip/actions/runs/14366609969/job/40282046693

```
2025-04-09T21:44:28.9642780Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/MTRPerControllerStorageTests.m:378: error: -[MTRPerControllerStorageTests test010_TestDataStoreMTRDeviceWithBulkReadWrite] : Asynchronous wait failed: Exceeded timeout of 30 seconds, with unfulfilled expectations: "Pairing Complete".
2025-04-09T21:45:08.1274690Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestControllerDelegate.mm:57: error: -[MTRPerControllerStorageTests test011_TestDataStoreMTRDeviceWithStorageBehaviorOptimizationDisabled] : ((error.code) equal to (0)) failed: ("9") is not equal to ("0")
2025-04-09T21:45:23.9411610Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test012_startSuspended] : failed - LEAKS DETECTED
2025-04-09T21:45:35.7110910Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test013_suspendDevices] : failed - LEAKS DETECTED
2025-04-09T21:45:45.8247340Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test014_TestDataStoreMTRDeviceInvalidateFlush] : failed - LEAKS DETECTED
2025-04-09T21:45:53.1507290Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testControllerServer] : failed - LEAKS DETECTED
2025-04-09T21:46:06.8293860Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingAttributes] : failed - LEAKS DETECTED
2025-04-09T21:46:14.4433790Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingClusters] : failed - LEAKS DETECTED
2025-04-09T21:46:23.2670650Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingEndpoints] : failed - LEAKS DETECTED
2025-04-09T21:46:23.9723080Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/MTRPerControllerStorageTests.m:3697: error: -[MTRPerControllerStorageTests testMTRDeviceDealloc] : ((controller.devices.count) equal to (0)) failed: ("1") is not equal to ("0")
2025-04-09T21:46:59.2161930Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestControllerDelegate.mm:57: error: -[MTRPerControllerStorageTests testMTRDeviceMaybeUnreachableIgnoredIfReceivingFromDevice] : ((error.code) equal to (0)) failed: ("9") is not equal to ("0")
2025-04-09T21:47:34.1234000Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestControllerDelegate.mm:57: error: -[MTRPerControllerStorageTests testMTRDeviceResetSubscription] : ((error.code) equal to (0)) failed: ("9") is not equal to ("0")
2025-04-09T21:47:44.3395890Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testSetMRPParametersWithRunningController] : failed - LEAKS DETECTED
2025-04-09T21:48:19.8762660Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testSubscriptionPool] : failed - LEAKS DETECTED
2025-04-09T21:51:16.2103020Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testSubscriptionPoolManyDevices] : failed - LEAKS DETECTED
2025-04-09T21:51:54.9953000Z ** TEST FAILED **
```

https://github.com/project-chip/connectedhomeip/actions/runs/14371491151/job/40296071813

```
2025-04-10T04:08:38.7082660Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/MTRPerControllerStorageTests.m:378: error: -[MTRPerControllerStorageTests test005_TestSessionResumptionDataClearingNodeIDChanged] : Asynchronous wait failed: Exceeded timeout of 30 seconds, with unfulfilled expectations: "Pairing Complete".
2025-04-10T04:09:14.6523510Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestControllerDelegate.mm:57: error: -[MTRPerControllerStorageTests test006_TestSessionResumptionDataClearingCATsChanged] : ((error.code) equal to (0)) failed: ("9") is not equal to ("0")
2025-04-10T04:09:22.5660470Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test007_TestMultipleControllers] : failed - LEAKS DETECTED
2025-04-10T04:09:26.0081170Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test008_TestDataStoreDirect] : failed - LEAKS DETECTED
2025-04-10T04:09:51.3443380Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test009_TestDataStoreMTRDevice] : failed - LEAKS DETECTED
2025-04-10T04:10:16.3070910Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test010_TestDataStoreMTRDeviceWithBulkReadWrite] : failed - LEAKS DETECTED
2025-04-10T04:10:25.6657720Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test011_TestDataStoreMTRDeviceWithStorageBehaviorOptimizationDisabled] : failed - LEAKS DETECTED
2025-04-10T04:10:32.9309430Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test012_startSuspended] : failed - LEAKS DETECTED
2025-04-10T04:10:39.8919260Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test013_suspendDevices] : failed - LEAKS DETECTED
2025-04-10T04:10:47.2941650Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests test014_TestDataStoreMTRDeviceInvalidateFlush] : failed - LEAKS DETECTED
2025-04-10T04:10:51.5261520Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testControllerServer] : failed - LEAKS DETECTED
2025-04-10T04:10:57.8729160Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingAttributes] : failed - LEAKS DETECTED
2025-04-10T04:11:03.9462010Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingClusters] : failed - LEAKS DETECTED
2025-04-10T04:11:10.1311960Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingEndpoints] : failed - LEAKS DETECTED
2025-04-10T04:11:14.1962900Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testMTRDeviceDealloc] : failed - LEAKS DETECTED
2025-04-10T04:11:20.0186730Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testMTRDeviceMaybeUnreachableIgnoredIfReceivingFromDevice] : failed - LEAKS DETECTED
[... cut ...] ] : failed - LEAKS DETECTED
2025-04-10T04:12:00.9939140Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testSubscriptionPool] : failed - LEAKS DETECTED
2025-04-10T04:14:22.6712200Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:120: error: -[MTRPerControllerStorageTests testSubscriptionPoolManyDevices] : failed - LEAKS DETECTED
```

andy31415, 2025-04-10:
Guessing ` Asynchronous wait failed: Exceeded timeout of 30 seconds, with unfulfilled expectations: "Pairing Complete".` triggers all the rest ...

bzbarsky-apple, 2025-04-10:
> https://github.com/project-chip/connectedhomeip/actions/runs/14371491151/job/40296071813

This looks like it's failing with:

```
2025-04-10T04:08:08.1343650Z 2025-04-10 04:08:08.012871+0000 xctest[27149:97862] [Discovery] Mdns: OnGetAddrInfo flags: 1073741827, interface: 4294967295, hostname: 6200719C66C4.local.
2025-04-10T04:08:08.1347090Z 2025-04-10 04:08:08.012942+0000 xctest[27149:97862] [Discovery] Mdns: OnNewAddress instance: 211D5B0789241CB2-0000000000000011 interface: 4294967295 ip: fe80::14de:bd3e:e872:a0c5
2025-04-10T04:08:08.1349270Z 2025-04-10 04:08:08.013505+0000 xctest[27149:97862] [Discovery] Mdns: Ignoring link-local address with no usable interface
2025-04-10T04:08:08.1350910Z 2025-04-10 04:08:08.013632+0000 xctest[27149:97862] [Discovery] Mdns: OnGetAddrInfo flags: 1073741827, interface: 1, hostname: 6200719C66C4.local.
2025-04-10T04:08:08.1352800Z 2025-04-10 04:08:08.013847+0000 xctest[27149:97862] [Discovery] Mdns: OnGetAddrInfo flags: 1073741826, interface: 1, hostname: 6200719C66C4.local.
2025-04-10T04:08:08.1355640Z 2025-04-10 04:08:08.013974+0000 xctest[27149:97862] [Discovery] Mdns: Resolve failure (kDNSServiceErr_BadState)
```

In the cases that succeed we get `::1` as an address, but this one for some reason only came back with an unusable link-local....  And yes, after that the rest is consequences from this failure.

> https://github.com/project-chip/connectedhomeip/actions/runs/14366609969/job/40282046693

Same deal:

```
2025-04-09T21:43:57.7512860Z 2025-04-09 21:43:57.684357+0000 xctest[33265:121336] [Discovery] Mdns: OnGetAddrInfo flags: 1073741827, interface: 4294967295, hostname: CA1BE81CC45A.local.
2025-04-09T21:43:57.7513800Z 2025-04-09 21:43:57.684406+0000 xctest[33265:121336] [Discovery] Mdns: OnNewAddress instance: CFC205C2DC1FFEEE-0000000000000011 interface: 4294967295 ip: fe80::411:8eb6:bde6:31bb
2025-04-09T21:43:57.7514410Z 2025-04-09 21:43:57.684429+0000 xctest[33265:121336] [Discovery] Mdns: Ignoring link-local address with no usable interface
2025-04-09T21:43:57.7515120Z 2025-04-09 21:43:57.684448+0000 xctest[33265:121336] [Discovery] Mdns: OnGetAddrInfo flags: 1073741827, interface: 1, hostname: CA1BE81CC45A.local.
2025-04-09T21:43:57.7515810Z 2025-04-09 21:43:57.684470+0000 xctest[33265:121336] [Discovery] Mdns: OnGetAddrInfo flags: 1073741826, interface: 1, hostname: CA1BE81CC45A.local.
2025-04-09T21:43:57.7516300Z 2025-04-09 21:43:57.684495+0000 xctest[33265:121336] [Discovery] Mdns: Resolve failure (kDNSServiceErr_BadState)
```

Clearly OnGetAddrInfo is being called with an error, but our logs don't say what error...  Will add more logging.

bzbarsky-apple, 2025-04-10:
https://github.com/project-chip/connectedhomeip/pull/38351 to make the logging better, then we can try to reproduce this again....

bzbarsky-apple, 2025-04-11:
> Clearly OnGetAddrInfo is being called with an error

Or actually... it's being called with an interface that we did not get anything useful on before (interface 1), so we ignore those results.

PR #38477 (2025-04-25): [Matter.framework][xctests] Ensure xctests advertising the controller…
… advertise locally

The `xctests` keeps advertising on interface `0` - which is basically all interfaces. This seems to confused `mdns`.

Fix #38341

#### Testing
CI is running - hopefully it will reduce the flakiness.
closed 2025-04-25 by mergify[bot]