project-chip/connectedhomeip #38402: [BUG] iOS bug while adding Matter device to custom fabric

sudhanshgupta, 2025-04-15:
### Reproduction steps

While implementing Matter Device commissioning into custom fabric, we observed following bug in iOS:

**Steps to reproduce:**

1. We’ve set the ecosystem name as "**Custom Fabric**" while calling **request.perform()**, referred:

[Matter Add Device Request](https://developer.apple.com/documentation/mattersupport/matteradddevicerequest/perform())
[Matter Add Device Request Topology](https://developer.apple.com/documentation/mattersupport/matteradddevicerequest/topology-swift.struct)

2. Intermediate UI are shown from Matter Support framework where in last pop up says **Accessory added to "Custom Fabric"**

3. Matter Extension receives the **onboardingPayload (MTRPayload)** and then app proceeds commissioning into custom fabric as a final step.

```
/// Commissions the device with the onboarding payload.
     
/// - Parameters:
///   - home: The selected home for the device.
///   - onboardingPayload: The onboarding payload, as defined by Matter specification, that you need to use to commission the device.
///   - commissioningID: A generated identifier you use with other MatterSupport methods.
    
open func commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) async throws
```

**Bug:** 
Refer to the steps, commissioning is happening in Step 3, but matter support framework pops up UI at Step 2 that displays the message "Accessory added..."

This pop up is misleading the user, please fix this.

**Device:**
Tapo Mini Plug

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

NA

### Platform

darwin

### Platform Version(s)

iOS

### Anything else?

[image]

[image]

[image]

bzbarsky-apple, 2025-04-21:
@sudhanshgupta This issue seems to be about the AppleSupport framework, not the Matter SDK.  Please see https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple for instructions on where and how to file it.

closed 2025-04-21 by bzbarsky-apple

sudhanshgupta, 2025-04-24:
@bzbarsky-apple  Thanks for the update.
We already raised feedback ticket: https://feedbackassistant.apple.com/feedback/17236444 last week for this with logs and waiting for response from Apple Team.
Is there any way to expedite on this so to resolve this issue sooner?

bzbarsky-apple, 2025-04-24:
I don't have a way to expedite anything there, particularly, not my team.  But I would have expected you to have heard back from them by now....