project-chip/connectedhomeip #38425: Python Controller Stalls Due to Missing Timeout for PASE Session.

Saravana-kr22, 2025-04-16:
**Description:**

While executing the TC_Darwin_pair.py script as part of the stress testing process, we observed a failure during the Establish PASE session phase using the Python controller. The error encountered was: `PASESession timed out while waiting for a response from the peer`.

This indicates that the PASE (Password Authenticated Session Establishment) session failed due to a timeout, implying that the controller did not receive a response from the peer device within the expected timeframe.

However, instead of terminating the session after the specified timeout duration, the controller continued to wait indefinitely with the following log message: `Waiting to discover commissionees that match our filters`.

This behavior is inconsistent with the expected outcome defined in the specification, which mandates that the session should time out after 60 seconds of inactivity or no response. The prolonged wait causes the test case to remain blocked during the Establish PASE session step, thereby delaying the entire testing process unnecessarily.

**controller API used for pase session** - `await self.th1.EstablishPASESession(str(self.manual_code), 
                                                                   self.dut_node_id)`

GitHub hash of the SDK that were being used are : 0b33b64c9ea4f72875217aace497b226d9dfac65

**Platforms**
I Have performed the execution on Raspberrypi as DUT using on-network provisioning

**Test Bed :**
Controller Platform - python controller running on  Raspberrypi 4B -8GB.
Raspberrypi as DUT - all-cluster-app running on Raspberrypi 4B -8GB.
Provisioning method - on-network

**Test scenario**
Pair/Provision the DUT (All-cluster-app running on raspi) as per drawiin pairing flow using on-network.
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
Factory reset the DUT
repeat step1 - 6 for 500 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Darwin_Pair.py

**Error logs:**

[controller_log_iteration_1_2025-04-15_21-28-33.log](https://github.com/user-attachments/files/19776431/controller_log_iteration_1_2025-04-15_21-28-33.log)

[Dut_log_iteration_1_2025-04-15T21_28_33_997258.log](https://github.com/user-attachments/files/19776442/Dut_log_iteration_1_2025-04-15T21_28_33_997258.log)

Harshith-GRL, 2025-04-22:
hey @bzbarsky-apple we are seeing this issue quite often from last week the code execution is getting stuck when calling the piece of code` await self.th1.EstablishPASESession(str(self.manual_code),  self.dut_node_id)`,
 in the terminal we can see a message `Waiting to discover commissionees that match our filters` and no further movement in the execution

Harshith-GRL, 2025-04-22:
@bzbarsky-apple here is a screenshot from the terminal hope you will be able to better understand the problem, observe the timestamps on the log information of the screenshot

[image]

bzbarsky-apple, 2025-04-23:
@Harshith-GRL I am not asking you about your problem.  I am asking you about where your claim:

> This behavior is inconsistent with the expected outcome defined in the specification, which mandates that the session should time out after 60 seconds of inactivity or no response

comes from.  As in, you are claiming the Python controller is violating the spec.  Which exact part of the spec is it violating?

And if it's NOT violating the spec, why do you think the issue is with the controller and not the way it's being used or the server side?

Saravana-kr22, 2025-04-23:
@bzbarsky-apple The spec explicitly mandates If the Commissioner (controller) does not send PAKE3 within 60 seconds, the Commissionee must actively terminate the session.

> 2025-04-15 13:57:33,600 - chip.native.CTL - ERROR - Commissioning discovery over BLE failed: Timeout
2025-04-15 13:57:33,601 - chip.native.- - ERROR - src/platform/Linux/BLEManagerImpl.cpp:740: CHIP Error 0x00000032: Timeout
2025-04-15 13:57:33,610 - chip.native.BLE - INFO - Scan complete. No matching device found.

From this log message it seems like controller could not discover the Commissionee (DUT) over BLE.

> 2025-04-15 13:58:01,566 - chip.native.SC - ERROR - PASESession timed out while waiting for a response from the peer. Expected message type was 64

This indicate that PASE handshake started but stalled and PAKE3 was never reviced

> 2025-04-15 13:58:01,569 - chip.native.CTL - INFO - Ignoring SecurePairingFailed status for now; we are waiting to see if we discover more devices
2025-04-15 13:58:01,570 - chip.native.CTL - INFO - Waiting to discover commissionees that match our filters

This log indicate that the controller ignored the PASE timeout and kept retrying indefinitely.

Spec refeence : https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/CommissioningFlows.adoc

[image]

bzbarsky-apple, 2025-04-24:
> The spec explicitly mandates If the Commissioner (controller) does not send PAKE3 within 60 seconds, the Commissionee must actively terminate the session.

Sure, but the Python thing is the _commissioner_, not the _commissionee_.

bzbarsky-apple, 2025-04-24:
And also: that bit of the spec being quoted only applies once the PASE handshake has started.  In this case it's not even starting, yes?

Saravana-kr22, 2025-04-25:
@bzbarsky-apple currently we are facing an issue during the PASE session establishment process. When the controller attempts to establish a session, it continues to wait indefinitely even after encountering the following error:

> 2025-04-15 13:57:33,600 - chip.native.CTL - ERROR - Commissioning discovery over BLE failed: Timeout
2025-04-15 13:57:33,601 - chip.native.- - ERROR - src/platform/Linux/BLEManagerImpl.cpp:740: CHIP Error 0x00000032: Timeout

Based on my understanding, once this timeout error occurs, the EstablishPASESession API should raise an exception. However, in this case, the call seems to hang without raising any exception, which blocks the subsequent iterations to be exexcuted. This behavior is proving to be a blocker for stress test runs.

Here is the relevant line of code being used:
`self.th1_pasesession = await self.th1.EstablishPASESession(str(self.manual_code),  self.dut_node_id)`

Could you please help confirm whether this issue lies within the Python API itself or the way the API is being invoked? Additionally, any suggestions on where this should be addressed would be more helpful.

bzbarsky-apple, 2025-04-26:
OK, so if we are talking about a practical problem not a spec-compliance one...

If this thing is using SetUpCodePairer, which it sounds like it is, based on that log, what I would expect to happen is:

1. There is a 30-second timer that terminates discovery.
2. Once that fires, any existing discovered things will attempt be contacted, then the whole thing should fail.

In this log, it looks like we think we still have discovery running after the "terminates discovery" bits.  Looks like that's due to this:

```
2025-04-15 13:57:13,577 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:41 - WiFi-PAF: has RendezvousInformationFlag::kWiFiPAF
2025-04-15 13:57:13,578 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:41 - Starting commissioning discovery over WiFiPAF
2025-04-15 13:57:13,579 - chip.native.CTL - ERROR - __init__.py:_RedirectToPythonLogging:39 - Error, Long discriminator is required
```

and then that never gets stopped.  As in, WiFiPAF discovery state management is broken....   

Fix coming up.

closed 2025-04-28 by mergify[bot]