project-chip/connectedhomeip #38490: [Feature] Support Model Context Protocol (MCP) Integration

briliant-ben, 2025-04-19:
### Feature description

## Motivation

Model Context Protocol (MCP) is very popular at present. The MCP client and MCP server architecture proposed by it allows access to resources and tools of local and private devices, which greatly enhances the functionality of LLM. 

The Matter protocol unifies the interconnection protocols between IOT devices and establishes interactive data model for device control and resource acquisition.

MCP unifies the communication between devices and LLM, while Matter unifies the communication between different types of devices. So, if the two can be integrated, then LLM can realize intelligent control of IOT devices through MCP and Matter protocols.

## Integration and Bridging

MCP communicates with LLM via JSON-RPC, and this design is also used in the matter implementation (e.g. [CHIPClusters.py](https://github.com/project-chip/connectedhomeip/blob/master/src/controller/python/chip/clusters/CHIPClusters.py)).

So, it is critical to **establish a new protocol** between LLM and Matter interactive data models, as a bridge between MCP and Matter.

## Implementation Considerations

- The matter protocol currently focuses on the control of IOT devices and is not comprehensive in terms of resource acquisition (including sound, images, videos, etc.). The current LLM is better at data resource analysis, but not at real-time control of devices.
- The MCP protocol is very new, and long-term changes may be very large.
- LLM illusion on outputs contradict the deterministic inputs of the Matter protocol.

## Proposal

- Open discussion on how such a protocol could be introduced in alignment with Matter architecture.
- Evaluate the feasibility of creating a working group or feature branch to prototype MCP support.

Please let me know if you'd like.

### Platform

all

### Platform Version(s)

N/A

### Anything else?

## Related Work

I have implemented a simple project for demonstration purposes. [MCP_over_Matter](https://github.com/briliant-ben/MCP_over_Matter). 

[image]

seriousme, 2025-05-29:
I just asked Gemini Flash 2.5 and it came up with the following (based on @matter/node.js):

---
Here's a conceptual outline and a more detailed implementation for an MCP server that acts as a Matter Controller, exposing Matter devices as MCP resources and tools.

### Core Idea:

- MCP Server: This will be your Node.js application, using @modelcontextprotocol/sdk to define resources and tools.
- Matter Controller: Inside the MCP server, you'll instantiate and manage a @matter/controller instance to interact with the Matter network.
- Integration Points:
  - Matter Device Discovery (MCP Resource): An MCP resource will allow the LLM to "ask" for a list of all currently commissioned Matter devices.
  - Device Status (MCP Resource): For each Matter device, you'll likely expose its attributes (e.g., on/off state, brightness, temperature) as an MCP resource that the LLM can read.
  - Device Control (MCP Tool): An MCP tool will allow the LLM to send commands to Matter devices (e.g., turn light on/off, set brightness, change thermostat temperature).

### Architecture:
```
+---------------------+           +---------------------+
|                     |           |                     |
|    AI Model / LLM   |<--------->|    MCP Client       |
|                     |           | (e.g., Claude App)  |
+---------------------+           +---------------------+
           |                             |
           | MCP (HTTP POST /mcp)        |
           V                             V
+-------------------------------------------------------+
|                                                       |
|              Node.js MCP Server                       |
|   (using @modelcontextprotocol/sdk & Express.js)      |
|                                                       |
|   +-----------------------------------------------+   |
|   |          MatterController Integration         |   |
|   | (using @matter/nodejs, @matter/controller)    |   |
|   |                                               |   |
|   |  - Manages Matter Fabric / Device State       |   |
|   |  - Handles Matter Device Discovery (mDNS)     |   |
|   |  - Executes Matter Commissioning              |   |
|   |  - Reads/Writes Matter Attributes             |   |
|   |  - Sends Matter Commands                      |   |
|   +-----------------------------------------------+   |
|                                                       |
|   MCP Definitions:                                    |
|   - Resource: `matter://devices` (list all devices)   |
|   - Resource: `matter://device/{nodeId}/status`       |
|                 (get device attributes/status)        |
|   - Tool:     `matter-control-device` (send commands) |
|                                                       |
+-------------------------------------------------------+
           |
           | Matter Protocol (UDP, mDNS, TCP)
           V
+---------------------+   +---------------------+   +---------------------+
|                     |   |                     |   |                     |
|   Matter Device 1   |<->|   Matter Device 2   |<->|   Matter Device N   |
|   (e.g., Light)     |   | (e.g., Thermostat)  |   | (e.g., Plug)        |
+---------------------+   +---------------------+   +---------------------+
```
---
The question would be if this should be part of connectedhomeip or a separate project glueing MCP to Matter?  

Hope this helps!
Hans

briliant-ben, 2025-05-30:
@seriousme ，THS for your response, I think it should be a separate project but controlled by connectedhomeip. However still no official response, maybe they don't think about it for now.  Hope someone could push this issue.