project-chip/connectedhomeip #38538: [CERT-TEST-FAILURE] [TC-CADMIN-1.5] Python script is failing at step-3

KishokG, 2025-04-23:
### Feature Area

Other

### Test Case

TC-CADMIN-1.5

### Reproduction steps

- **Command used to launch the DUT :** ./chip-all-clusters-app
- **Command used to run the python script :** python3 TC_CADMIN_1_5.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json
- While validating the TC-CADMIN-1.5 Python script, the test case is failing at Step 3, where TH_CR1 is expected to find DUT_CE advertising as a commissionable node on DNS-SD. The expected outcome is to verify that CM=2.
- However, the issue seems to be that in Step 3, the script does not filter using the discriminator value that was used in Step 2 when opening the commissioning window. This leads to a mismatch and causes the failure.

**For example, the DUT received the open commissioning window command from TH with discriminator as 2601**

[image]

**But in step 3 the TH is checked the CM value in discriminator 2795**

[image]

**Here is the failure log files for reference:** 
[CADMIN-1.5_failure logs.txt](https://github.com/user-attachments/files/19863974/CADMIN-1.5_failure.logs.txt)
[CADMIN-1.5_failure logs_DUT.txt](https://github.com/user-attachments/files/19863973/CADMIN-1.5_failure.logs_DUT.txt)

### Bug prevalence

Whenever I'm executing this TC

### GitHub hash of the SDK that was being used

ff5490ab54b9e68746bdc5ea853a08a0425be1f2

### Platform

raspi

j-ororke, 2025-04-24:
Provided suggested fix in PR #[38573](https://github.com/project-chip/connectedhomeip/pull/38573)
- This suggested fix should resolve the issue of discovering the correct commissionable device in a congested environment where multiple devices could be commissionable.

KishokG, 2025-04-28:
Issue resolved

closed 2025-04-28 by KishokG