project-chip/connectedhomeip #38556: [CERT-TEST-FAILURE] [TC-ACE-2.3] Python script is failing in step 3b while running using All-Clusters-app

KishokG, 2025-04-24:
### Feature Area

Other

### Test Case

TC-ACE-2.3

### Reproduction steps

- **Build the sample app using:** scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ examples/all-clusters-app/linux/out/all-clusters-app chip_inet_config_enable_ipv4=false
- **Advertise the DUT(All-Clusters-app) by sending command:** ./chip-all-clusters-app
- **Build the python environment using:** scripts/build_python.sh -m platform -d true -i python_env
- **Activate the python environment:** source/python_env/bin/activate
- **Run the python script:** python3 TC_AccessChecker.py  --commissioning-method on-network --discriminator 3840 --passcode 20202021  --storage-path admin_storage.json   --tests test_TC_ACE_2_3 --timeout 1000

**Summary of Errors for ACE-2.3**

>                  ******************************************************************
>                  *
>                  * Test test_TC_ACE_2_3 failed for the following reason:
>                  * Details=One or more access violations was found, Extras=None
>                  *
>                  * File "/home/ubuntu/Requestor/connectedhomeip/no/lib/python3.12/site-packages/chip/testing/basic_composition.py", line 214, in fail_current_test
>                  * asserts.fail(msg)
>                  *
>                  * Test step:
>                  *     3b_kAdminister: For each standard command on each standard cluster on each endpoint,
> TH_second_controller checks the permission requirements for that command.
> If the permission required for the command is HIGHER than the permission level being tested,
> TH_second_controller sends the command to the DUT using default values.
> Regardless of the command contents, the DUT should return an access error since access must be checked
> before the command is processed. Receipt of an UNSUPPORTED_COMMAND error is a conformance failure.	Expected outcome: DUT returns UNSUPPORTED_ACCESS error
>                  *
>                  * Endpoint: None
>                  *
>                  *******************************************************************
> 
> [MatterTest] 04-24 06:30:08.183 INFO [Test] test_TC_ACE_2_3 FAIL
> [MatterTest] 04-24 06:30:08.191 INFO ###########################################################
> [MatterTest] 04-24 06:30:08.191 INFO Problems found:
> [MatterTest] 04-24 06:30:08.191 INFO ===============
> [MatterTest] 04-24 06:30:08.192 INFO 
> Problem: ProblemSeverity.WARNING
>     test_name: Parse Device Type XML
>     location: 
>        DeviceType: 144
>     problem: Unknown cluster id ID-TBD
>     spec_location: 
> 
> [MatterTest] 04-24 06:30:08.192 INFO 
> Problem: ProblemSeverity.WARNING
>     test_name: Parse Device Type XML
>     location: 
>        DeviceType: 145
>     problem: Unknown cluster id ID-TBD
>     spec_location: 
> 
> [MatterTest] 04-24 06:30:08.192 INFO 
> Problem: ProblemSeverity.WARNING
>     test_name: Access Checker
>     location: 
>        Endpoint: 0,
>        Cluster:  769 (0x301) BallastConfiguration
>     problem: Cluster not present in spec data
>     spec_location: 
> 
> [MatterTest] 04-24 06:30:08.193 INFO 
> Problem: ProblemSeverity.ERROR
>     test_name: Command test - privilege 3
>     location: 
>        Endpoint: 0,
>        Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor
>        Command:  0 (0x00)
>     problem: Unexpected error sending command <class 'chip.clusters.Objects.OtaSoftwareUpdateRequestor.Commands.AnnounceOTAProvider'> with privilege 3 - expected UNSUPPORTED_ACCESS, got 129
>     spec_location: 
> 
> [MatterTest] 04-24 06:30:08.193 INFO 
> Problem: ProblemSeverity.ERROR
>     test_name: Command test - privilege 4
>     location: 
>        Endpoint: 0,
>        Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor
>        Command:  0 (0x00)
>     problem: Unexpected error sending command <class 'chip.clusters.Objects.OtaSoftwareUpdateRequestor.Commands.AnnounceOTAProvider'> with privilege 4 - expected UNSUPPORTED_ACCESS, got 129
>     spec_location: 

**Log file for reference:** [TC-ACE-2.3.txt](https://github.com/user-attachments/files/19885884/TC-ACE-2.3.txt)

### Bug prevalence

Whenever I'm executing this TC

### GitHub hash of the SDK that was being used

f647fda9ee8167c383e00ed9a911c4c7645e56b8

### Platform

raspi

cecille, 2025-04-28:
That's the case of a test finding a real problem - the linux version of most example apps do not properly set up the OTA provider cluster. The all clusters app claims support for that command in its accepted command list, but attempts to invoke it return UNSUPPORTED_COMMAND.

cecille, 2025-04-28:
In the CI, we run this with --bool-arg ci_only_linux_skip_ota_cluster_disallowed_for_certification:True, but note that as the flag states, it's disallowed for certification.

cecille, 2025-04-28:
Issue for OTA here: https://github.com/project-chip/connectedhomeip/issues/36716, but we don't really have a good solution given that the zap file is common.

KishokG, 2025-04-28:
Thanks @cecille. Using the mentioned argument, the test cases pass when running with the All-Clusters-App; however, it is disallowed for certification. Since this is a sample app issue, we will move this test case to TE2.

closed 2025-04-28 by KishokG