project-chip/connectedhomeip #38580: [BUG] TCP large message Session .HasValue() in Energy Gateway App

jamesharrow, 2025-04-24:
### Reproduction steps

I'm not 100% sure if this is my doing, but don't think it is.

When running chip-energy-gateway-app with TC_SEPR_2_3.py, it crashes out just after commissioning (step 3) of the test case tries to send GetDetailedForecastRequest command (which is a 'largeMessage'). It never reaches the cluster handler code 'InvokeCommand' and Segmentation Fault happens.

1. build chip-energy-gateway-app for Linux
2. run it with test enable triggers enabled: 
```
rm /tmp/chip*; valgrind --leak-check=full --show-leak-kinds=all --track-origins=yes out/debug/chip-energy-gateway-app --enable-key 000102030405060708090a0b0c0d0e0f
```
3. ensure have built chip-repl: 
 ```./scripts/build_python.sh -i out/separate```, then ```. out/separate/bin/activate.sh```
4. Manually run python script TC_SEPR_2_3.py:`
```python src/python_testing/TC_SEPR_2_3.py -m on-network -p 20202021 -d 3840 --hex-arg enableKey:000102030405060708090a0b0c0d0e0f --endpoint 1```

It fails at step 3 (the energy-gateway-app crashes):

Application logs:
```
1745533342.657] [2984609:2984609] [DMG] Received command for Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0004
[1745533342.664] [2984609:2984609] [FS] GeneralCommissioning: Received CommissioningComplete
[1745533342.677] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.686] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.688] [2984609:2984609] [FP] Metadata for Fabric 0x1 persisted to storage.
[1745533342.702] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.716] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.721] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.725] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.736] [2984609:2984609] [TS] Committing Last Known Good Time to storage: 2023-10-14T01:16:48
[1745533342.740] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.763] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.777] [2984609:2984609] [ZCL] OpCreds: Fabric index 0x1 was committed to storage. Compressed Fabric Id 0xCE87870D6EA3A122, FabricId 0000000000000001, NodeId 0000000012344321, VendorId 0xFFF1
[1745533342.811] [2984609:2984609] [DL] Wrote settings to /tmp/chip_kvs
[1745533342.813] [2984609:2984609] [FS] GeneralCommissioning: Successfully committed pending fabric data
[1745533342.814] [2984609:2984609] [FS] Fail-safe cleanly disarmed
[1745533342.816] [2984609:2984609] [DMG] Endpoint 0, Cluster 0x0000_0030 update version to fc98d51
[1745533342.817] [2984609:2984609] [DMG] Command handler moving to [NewRespons]
[1745533342.818] [2984609:2984609] [DMG] Command handler moving to [ Preparing]
[1745533342.818] [2984609:2984609] [DMG] Command handler moving to [AddingComm]
[1745533342.820] [2984609:2984609] [DMG] Command handler moving to [AddedComma]
[1745533342.824] [2984609:2984609] [DMG] Decreasing reference count for CommandHandlerImpl, remaining 1
[1745533342.824] [2984609:2984609] [DMG] Decreasing reference count for CommandHandlerImpl, remaining 0
[1745533342.824] [2984609:2984609] [DMG] Command handler moving to [AwaitingDe]
[1745533342.825] [2984609:2984609] [EM] <<< [E:54297r S:22487 M:156431546 (Ack:201528961)] (S) Msg TX from 0000000012344321 to 1:000000000001B669 [A122] [UDP:[fd00::ef86:70ea:26ec:aba4]:56706] --- Type 0001:09 (IM:InvokeCommandResponse) (B:70)
[1745533342.826] [2984609:2984609] [EM] ??1 [E:54297r S:22487 M:156431546] (S) Msg Retransmission to 1:000000000001B669 scheduled for 376ms from now [State:Active II:500 AI:300 AT:4000]
[1745533342.827] [2984609:2984609] [DMG] Command response sender moving to [AllInvokeR]
[1745533342.828] [2984609:2984609] [SVR] Commissioning completed successfully
[1745533342.829] [2984609:2984609] [DIS] Updating services using commissioning mode 0
[1745533342.834] [2984609:2984609] [DIS] CHIP minimal mDNS started advertising.
[1745533342.888] [2984609:2984609] [DL] Using WiFi MAC for hostname
[1745533342.888] [2984609:2984609] [DIS] Advertise operational node CE87870D6EA3A122-0000000012344321
[1745533342.889] [2984609:2984609] [DIS] Responding with _matter._tcp.local
[1745533342.889] [2984609:2984609] [DIS] Responding with CE87870D6EA3A122-0000000012344321._matter._tcp.local
[1745533342.889] [2984609:2984609] [DIS] Responding with CE87870D6EA3A122-0000000012344321._matter._tcp.local
[1745533342.889] [2984609:2984609] [DIS] Responding with 080027E6F601.local
[1745533342.890] [2984609:2984609] [DIS] Responding with _ICE87870D6EA3A122._sub._matter._tcp.local
[1745533342.890] [2984609:2984609] [DIS] CHIP minimal mDNS configured as 'Operational device'; instance name: CE87870D6EA3A122-0000000012344321.
[1745533342.899] [2984609:2984609] [DIS] mDNS service published: _matter._tcp
[1745533342.900] [2984609:2984609] [DL] Using WiFi MAC for hostname
[1745533342.904] [2984609:2984609] [DIS] Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=0 cp=0
[1745533342.905] [2984609:2984609] [DIS] Responding with _matterc._udp.local
[1745533342.905] [2984609:2984609] [DIS] Responding with 7E03DF447D7CAF7A._matterc._udp.local
[1745533342.905] [2984609:2984609] [DIS] Responding with 080027E6F601.local
[1745533342.905] [2984609:2984609] [DIS] Responding with _V65521._sub._matterc._udp.local
[1745533342.905] [2984609:2984609] [DIS] Responding with _T1292._sub._matterc._udp.local
[1745533342.905] [2984609:2984609] [DIS] Responding with _S15._sub._matterc._udp.local
[1745533342.906] [2984609:2984609] [DIS] Responding with _L3840._sub._matterc._udp.local
[1745533342.906] [2984609:2984609] [DIS] Responding with 7E03DF447D7CAF7A._matterc._udp.local
[1745533342.906] [2984609:2984609] [DIS] CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 7E03DF447D7CAF7A.
[... 174 lines cut ...]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jebh/workspace/connectedhomeip/out/seperate/lib/python3.11/site-packages/chip/testing/matter_testing.py", line 1340, in send_single_cmd
    result = await dev_ctrl.SendCommand(nodeid=node_id, endpoint=endpoint, payload=cmd, timedRequestTimeoutMs=timedRequestTimeoutMs,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jebh/workspace/connectedhomeip/out/seperate/lib/python3.11/site-packages/chip/ChipDeviceCtrl.py", line 1430, in SendCommand
    return await future
           ^^^^^^^^^^^^
chip.exceptions.ChipStackError: src/app/CommandSender.cpp:344: CHIP Error 0x00000032: Timeout
[MatterTest] 04-24 23:22:25.004 INFO 		***** Test Failure : 
[MatterTest] 04-24 23:22:25.004 INFO Finished test in 2161556.0ms
[MatterTest] 04-24 23:22:25.005 ERROR 

```

### Bug prevalence

The command works if you run it manually from chip-repl (slowly) - see https://github.com/project-chip/connectedhomeip/issues/38575, but always crashes when run from TC_SEPR_2_3.py

### GitHub hash of the SDK that was being used

In PR# #38255 but master branch: bc7f2a4

### Platform

core

andy31415, 2025-04-25:
@jamesharrow could you run the application in an `-asan` variant or valgrind? It seems like a "use after free" to me.

bzbarsky-apple, 2025-04-25:
Looks like a null-deref to me...

jamesharrow, 2025-04-25:
I wanted to ensure it wasn't anything to do with my energy-gateway-app or the cluster code I'd written, so I have managed to reproduce this with all-clusters-app.

1. [OPTIONAL] Modify ```src/app/zap-templates/zcl/data-model/chip/onoff-cluster.xml``` to add '<quality largeMessage="true"/> to the Off and On commands.

```    <command source="client" code="0x00" name="Off" optional="false">
      <description>On receipt of this command, a device SHALL enter its ‘Off’ state. This state is device dependent, but it is recommended that it is used for power off or similar functions. On receipt of the Off command, the OnTime attribute SHALL be set to 0.</description>
      <quality largeMessage="true"/>
      <mandatoryConform/>
    </command>

    <command source="client" code="0x01" name="On" optional="false">
      <description>On receipt of this command, a device SHALL enter its ‘On’ state. This state is device dependent, but it is recommended that it is used for power on or similar functions. On receipt of the On command, if the value of the OnTime attribute is equal to 0, the device SHALL set the OffWaitTime attribute to 0.</description>
      <quality largeMessage="true"/>
      <mandatoryConform>
        <notTerm>
          <feature name="OFFONLY"/>
        </notTerm>
      </mandatoryConform>
    </command>
```

2. run regen-all

3.[QUICK WAY] - modify ```cluster-objects.cpp``` : 1398 to add:
```
bool CommandHasLargePayload(ClusterId aCluster, CommandId aCommand)
{
+    if ((aCluster == Clusters::OnOff::Id) && (aCommand == Clusters::OnOff::Commands::Off::Id))
+    {
+       return true;
+    }
+   if ((aCluster == Clusters::OnOff::Id) && (aCommand == Clusters::OnOff::Commands::On::Id))
+    {
+        return true;
+   }
...
    if ((aCluster == Clusters::CommodityPrice::Id) &&
        (aCommand == Clusters::CommodityPrice::Commands::GetDetailedForecastRequest::Id))
    {
```
4. Build all-clusters 'clang-asan': `./scripts/build/build_examples.py --target linux-x64-all-clusters-clang-asan build`
5. Run the app: `out/linux-x64-all-clusters-clang-asan/chip-all-clusters-app --enable-key 000102030405060708090a0b0c0d0e0f
`
6. (Using chip-repl - assuming you can build it with the instructions above) 
use the attached python file (which is a hacked version of TC_SEPR_2_3_hack.py) to change the command and cluster to be an OnOff command:

`python src/python_testing/TC_SEPR_2_3_hack.py -m on-network -p 20202021 -d 3840 --hex-arg enableKey:000102030405060708090a0b0c0d0e0f --endpoint 1`

```
[1745609907.836] [247140:247140] [DMG] Reading attribute: Cluster=0x0000_0033 Endpoint=0x0 AttributeId=0x0000_0008 (expanded=0)
[1745609907.836] [247140:247140] [DMG] <RE> Sending report (payload has 35 bytes)...
[1745609907.837] [247140:247140] [EM] <<< [E:35629r S:40611 M:17848040] (S) Msg TX from 0000000012344321 to 1:000000000001B669 [A122] [TCP:[fe80::a00:27ff:fee6:f601%enp0s3]:39260] --- Type 0001:05 (IM:ReportData) (B:65)
[1745609907.837] [247140:247140] [DMG] <RE> OnReportConfirm: NumReports = 0
[1745609907.837] [247140:247140] [DMG] <RE> ReportsInFlight = 0 with readHandler 0, RE has no more messages
[1745609907.837] [247140:247140] [DMG] IM RH moving to [AwaitingDestruction]
[1745609907.845] [247140:247140] [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1745609907.853] [247140:247140] [EM] >>> [E:35630r S:40611 M:213130211] (S) Msg RX from 1:000000000001B669 [A122] to 0000000012344321 --- Type 0001:0A (IM:TimedRequest) (B:39)
[1745609907.854] [247140:247140] [EM] Handling via exchange: 35630r, Delegate: 0x5610221b6268
[1745609907.856] [247140:247140] [DMG] TimedRequestMessage =
[1745609907.856] [247140:247140] [DMG] {
[1745609907.856] [247140:247140] [DMG] 	TimeoutMs = 0xbb8,
[1745609907.856] [247140:247140] [DMG] 	InteractionModelRevision = 12
[1745609907.856] [247140:247140] [DMG] }
[1745609907.856] [247140:247140] [DMG] Got Timed Request with timeout 3000: handler 0x70c92608e260 exchange 35630r
[1745609907.856] [247140:247140] [EM] <<< [E:35630r S:40611 M:17848041] (S) Msg TX from 0000000012344321 to 1:000000000001B669 [A122] [TCP:[fe80::a00:27ff:fee6:f601%enp0s3]:39260] --- Type 0001:01 (IM:StatusResponse) (B:38)
[1745609907.857] [247140:247140] [DMG] Timed Request time limit 0x0000000002174BA2: handler 0x70c92608e260 exchange 35630r
[1745609907.865] [247140:247140] [EM] >>> [E:35630r S:40611 M:213130212] (S) Msg RX from 1:000000000001B669 [A122] to 0000000012344321 --- Type 0001:08 (IM:InvokeCommandRequest) (B:59)
[1745609907.867] [247140:247140] [EM] Found matching exchange: 35630r, Delegate: 0x70c92608e260
[1745609907.867] [247140:247140] [DMG] Timed following action arrived at 0x0000000002173FF4: handler 0x70c92608e260 exchange 35630r
[1745609907.867] [247140:247140] [DMG] Handing timed invoke to IM engine: handler 0x70c92608e260 exchange 35630r
[... 143 lines cut ...]
    @run_if_endpoint_matches(has_cluster(cluster))
    async def test_TC_SEPR_2_3(self):
        """Run the test steps."""
        endpoint = self.get_endpoint()

        self.step("1")
        # Commission DUT - already done
        try:
            device = await self.default_controller.GetConnectedDevice(nodeid=self.dut_node_id, allowPASE=False, timeoutMs=1000,
                                                                      payloadCapability=ChipDeviceCtrl.TransportPayloadCapability.LARGE_PAYLOAD)
        except TimeoutError:
            asserts.fail("Unable to establish a CASE session over TCP to the device. Does the device support TCP?")

        asserts.assert_equal(device.sessionAllowsLargePayload, True, "Session does not have associated TCP connection")

        self.step("2")
        # TH reads TestEventTriggersEnabled attribute from General Diagnostics Cluster
        await self.check_test_event_triggers_enabled()

        # THIS CAUSES THE CRASH
        await self.send_long_onoff()

if __name__ == "__main__":
    default_matter_test_main()
```

bzbarsky-apple, 2025-04-26:
Yep, this is a null-deref:

```
frame #4: 0x000000010359ec2c chip-energy-gateway-app`chip::app::InteractionModelEngine::CheckCommandFlags(this=0x000000010385e240, aRequest=0x000000016d187a20, entry=0x000000016d1874a0) at InteractionModelEngine.cpp:1838:29
   1835	    // Command that is marked as having a large payload must be sent over a
   1836	    // session that supports it.
   1837	    if (entry.flags.Has(DataModel::CommandQualityFlags::kLargeMessage) &&
-> 1838	        !CurrentExchange()->GetSessionHandle()->AllowsLargePayload())
   1839	    {
   1840	        return Status::InvalidTransportType;
   1841	    }
(lldb) p CurrentExchange()
(chip::Messaging::ExchangeContext *) nullptr
```

Callstack:

```
  * frame #4: 0x000000010359ec2c chip-energy-gateway-app`chip::app::InteractionModelEngine::CheckCommandFlags(this=0x000000010385e240, aRequest=0x000000016d187a20, entry=0x000000016d1874a0) at InteractionModelEngine.cpp:1838:29
    frame #5: 0x000000010359d8b4 chip-energy-gateway-app`chip::app::InteractionModelEngine::ValidateCommandCanBeDispatched(this=0x000000010385e240, request=0x000000016d187a20) at InteractionModelEngine.cpp:1782:12
    frame #6: 0x0000000102fdb728 chip-energy-gateway-app`chip::app::CommandResponseSender::ValidateCommandCanBeDispatched(this=0x0000617000011f80, request=0x000000016d187a20) at CommandResponseSender.cpp:138:38
    frame #7: 0x0000000102f120f8 chip-energy-gateway-app`chip::app::CommandHandlerImpl::ProcessCommandDataIB(this=0x0000617000011fb0, aCommandElement=0x000000016d1889a0) at CommandHandlerImpl.cpp:402:45
    frame #8: 0x0000000102f0b61c chip-energy-gateway-app`chip::app::CommandHandlerImpl::ProcessInvokeRequest(this=0x0000617000011fb0, payload=0x000000016d18c940, isTimedInvoke=true) at CommandHandlerImpl.cpp:273:22
    frame #9: 0x0000000102f0a28c chip-energy-gateway-app`chip::app::CommandHandlerImpl::OnInvokeCommandRequest(this=0x0000617000011fb0, commandResponder=0x0000617000011f80, payload=0x000000016d18c940, isTimedInvoke=true) at CommandHandlerImpl.cpp:117:21
    frame #10: 0x0000000102fdbf08 chip-energy-gateway-app`chip::app::CommandResponseSender::OnInvokeCommandRequest(this=0x0000617000011f80, ec=0x000060c00007ed00, payload=0x000000016d18c940, isTimedInvoke=true) at CommandResponseSender.cpp:217:37
    frame #11: 0x0000000103589314 chip-energy-gateway-app`chip::app::InteractionModelEngine::OnInvokeCommandRequest(this=0x000000010385e240, apExchangeContext=0x000060c00007ed00, aPayloadHeader=0x000000016d18b930, aPayload=0x000000016d18c940, aIsTimedInvoke=true) at InteractionModelEngine.cpp:570:23
    frame #12: 0x00000001035a0d64 chip-energy-gateway-app`chip::app::InteractionModelEngine::OnTimedInvoke(this=0x000000010385e240, apTimedHandler=0x00006030000ba1f0, apExchangeContext=0x000060c00007ed00, aPayloadHeader=0x000000016d18b930, aPayload=0x000000016d18c940) at InteractionModelEngine.cpp:1932:21
    frame #13: 0x00000001035c4300 chip-energy-gateway-app`chip::app::TimedHandler::OnMessageReceived(this=0x00006030000ba1f0, aExchangeContext=0x000060c00007ed00, aPayloadHeader=0x000000016d18b930, aPayload=0x000000016d18c940) at TimedHandler.cpp:78:24
    frame #14: 0x0000000103319b24 chip-energy-gateway-app`chip::Messaging::ExchangeContext::HandleMessage(this=0x000060c00007ed00, messageCounter=137787635, payloadHeader=0x000000016d18b930, msgFlags=(mValue = 0), msgBuf=0x000000016d18c940) at ExchangeContext.cpp:617:31
```

The important part: this is NOT going through `InteractionModelEngine::OnMessageReceived`, and the fact that instead of passing the exchange through we are stashing it in some global on _some_ codepaths but not others is biting us.

closed 2025-04-27 by mergify[bot]