project-chip/connectedhomeip #38614: Use constexpr in PeerAddress

OlivierGre, 2025-04-25:
```suggestion
    static constexpr PeerAddress NFC(const uint16_t shortId) { return PeerAddress(Type::kNfc, shortId); }
```

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/37060#discussion_r2056950139_

OlivierGre, 2025-04-25:
Comment from @bzbarsky-apple:

Well, the constructor for PeerAddress should be constexpr too, right?

OlivierGre, 2025-05-20:
@bzbarsky-apple I have tried to do the change but I'm afraid I don't understand really the benefit of adding this constexpr keyword.
Shall I add it to all the constructors even the one with an IPAddress?

OlivierGre, 2025-05-21:
@bzbarsky-apple I have done the change but it is in fact not compatible with the [Issue 38615](https://github.com/project-chip/connectedhomeip/issues/38615) where we are changing the code to use a Union.

Here is the compilation error:
```
2025-05-21 10:53:08 INFO    ../../src/transport/raw/PeerAddress.h:249:15: error: constexpr constructor never produces a constant expression [-Winvalid-constexpr]
2025-05-21 10:53:08 INFO        constexpr PeerAddress(uint16_t shortId) : mTransportType(Type::kNfc) { mId.mNFCShortId = shortId; }
2025-05-21 10:53:08 INFO                  ^
2025-05-21 10:53:08 INFO    ../../src/transport/raw/PeerAddress.h:249:15: note: non-constexpr constructor 'Id' cannot be used in a constant expression
2025-05-21 10:53:08 INFO    ../../src/transport/raw/PeerAddress.h:263:11: note: declared here
2025-05-21 10:53:08 INFO        union Id
2025-05-21 10:53:08 INFO              ^
2025-05-21 10:53:08 INFO    1 error generated.
```

And here is the line of code:
`    constexpr PeerAddress(uint16_t shortId) : mTransportType(Type::kNfc) { mId.mNFCShortId = shortId; }`

Here is ChatGPT view:

> The compilation error you're encountering is related to the use of constexpr with a constructor that involves a union. In C++, constexpr constructors are intended to produce constant expressions, but there are certain restrictions, especially when dealing with unions.

So IMO, we can not use constexpr here.

OlivierGre, 2025-05-22:
Thank you @bzbarsky-apple .

`constexpr PeerAddress(uint16_t shortId) : mTransportType(Type::kNfc), mId{.mNFCShortId = shortId; } {}`
This line is causing a compilation error. I have removed the semi colon and the compilation is ok:
`constexpr PeerAddress(uint16_t shortId) : mTransportType(Type::kNfc), mId{.mNFCShortId = shortId} {}`

Do you agree with this change?

OlivierGre, 2025-05-22:
Fixed in https://github.com/project-chip/connectedhomeip/pull/37060.

bzbarsky-apple, 2025-05-22:
@OlivierGre Er, yes, the semicolon should not have been there.  Muscle-memory typo...

closed 2025-06-06 by OlivierGre