project-chip/connectedhomeip #38615: PeerAddress: Use an Union for mNFCShortId  and mRemoteId

OlivierGre, 2025-04-25:
I wonder whether it's worth putting mRemoteId and mNFCShortId in a union.  They're never used together, right?  Followup for that is best.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/37060#discussion_r2056950669_

OlivierGre, 2025-05-20:
@bzbarsky-apple In the union, it is not possible to set mRemoteId to 0.

Shall I set it to 0 in the constructors not using a remoteId parameter?
```
    PeerAddress() : mIPAddress(Inet::IPAddress::Any), mTransportType(Type::kUndefined) {}
    PeerAddress(const Inet::IPAddress & addr, Type type) : mIPAddress(addr), mTransportType(type) {}
    PeerAddress(Type type) : mTransportType(type) {}
    PeerAddress(Type type, NodeId remoteId) : mTransportType(type) { mId.mRemoteId = remoteId; }

    union Id
    {
        NodeId mRemoteId;
        uint16_t mNFCShortId;
    } mId;
```

OlivierGre, 2025-05-20:
Like this:
```
    PeerAddress() : mIPAddress(Inet::IPAddress::Any), mTransportType(Type::kUndefined) { mId.mRemoteId = 0; }
    PeerAddress(const Inet::IPAddress & addr, Type type) : mIPAddress(addr), mTransportType(type) { mId.mRemoteId = 0; }
    PeerAddress(Type type) : mTransportType(type) { mId.mRemoteId = 0; }
    PeerAddress(Type type, NodeId remoteId) : mTransportType(type) { mId.mRemoteId = remoteId; }
```

bzbarsky-apple, 2025-05-20:
@OlivierGre That looks fine, but I would consider setting to kUndefinedNodeId (which is in fact 0, but might be clearer in terms of what's being set).

OlivierGre, 2025-05-22:
Fixed in https://github.com/project-chip/connectedhomeip/pull/37060.

closed 2025-06-06 by OlivierGre