project-chip/connectedhomeip #38638: [CERT-TEST-FAILURE] [TC-AVSUM-2-7] Python script failure

Ashwinigrl, 2025-04-28:
### Feature Area

Other

### Test Case

TC-AVSUM-2-7

### Reproduction steps

1. **Build the sample app using:** `scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ examples/all-clusters-app/linux/out/all-clusters-app chip_inet_config_enable_ipv4=false`
2. **Advertise the DUT(All-Clusters-app) by sending command:** `./chip-all-clusters-app`
3. **Build the python environment using:** `scripts/build_python.sh -m platform -d true -i python_env`
4. **Activate the python environment:** `source/python_env/bin/activate`
5. **Run the python script:** `python3 TC_AVSUM_2_7.py --commissioning-method on-network --discriminator 3840 --endpoint 1 --passcode 20202021   --storage-path admin_storage.json`.

**Description of Issue:**

When we tried to execute the TC-AVSUM-2-7  python script is failing at test step 3 `Send DPTZSetVieport with an unknown stream ID, verify NotFound response` due to the Status value mismatch.

> MatterTest] 04-28 18:37:06.379 INFO 		***** Test Step 3 : Send DPTZSetVieport with an unknown stream ID, verify NotFound response
> [MatterTest] 04-28 18:37:06.380 INFO Found an existing secure session to [1:0000000012344321]!
> [MatterTest] 04-28 18:37:06.381 INFO <<< [E:34704i S:3770 M:62020392] (S) Msg TX from 000000000001B669 to 1:0000000012344321 [23BF] [UDP:[fe80::16fb:ddf1:36af:fdc5%wlp0s20f3]:5540] --- Type 0001:08 (IM:InvokeCommandRequest) (B:78)
> [MatterTest] 04-28 18:37:06.381 INFO ??1 [E:34704i S:3770 M:62020392] (S) Msg Retransmission to 1:0000000012344321 scheduled for 338ms from now [State:Active II:500 AI:300 AT:4000]
> [MatterTest] 04-28 18:37:06.381 INFO >>> [E:34704i S:3770 M:64108606 (Ack:62020392)] (S) Msg RX from 1:0000000012344321 [23BF] to 000000000001B669 --- Type 0001:09 (IM:InvokeCommandResponse) (B:68)
> [MatterTest] 04-28 18:37:06.384 INFO Received Command Response Status for Endpoint=1 Cluster=0x0000_0552 Command=0x0000_0005 Status=0x0
> [MatterTest] 04-28 18:37:06.384 ERROR Exception occurred in test_TC_AVSUM_2_7.
> Traceback (most recent call last):
>   File "/home/grl/connectedhomeip/no/lib/python3.12/site-packages/mobly/base_test.py", line 818, in exec_one_test
>     test_method()
>   File "/home/grl/connectedhomeip/no/lib/python3.12/site-packages/chip/testing/decorators.py", line 358, in per_endpoint_runner
>     test_instance.event_loop.run_until_complete(asyncio.wait_for(
>   File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
>     return future.result()
>            ^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
>     return await fut
>            ^^^^^^^^^
>   File "/home/grl/connectedhomeip/src/python_testing/TC_AVSUM_2_7.py", line 89, in test_TC_AVSUM_2_7
>     await self.send_dptz_set_viewport_command(endpoint, failingvideostreamID, smallfailingviewport, expected_status=Status.NotFound)
>   File "/home/grl/connectedhomeip/src/python_testing/TC_AVSUMTestBase.py", line 123, in send_dptz_set_viewport_command
>     asserts.assert_equal(expected_status, Status.Success)
>   File "/home/grl/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 66, in assert_equal
>     _call_unittest_assertion(
>   File "/home/grl/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 50, in _call_unittest_assertion
>     raise signals.TestFailure(my_msg, extras=extras)
> mobly.signals.TestFailure: Details=<Status.NotFound: 139> != <Status.Success: 0>, Extras=None
> [MatterTest] 04-28 18:37:06.385 INFO <<< [E:34704i S:3770 M:62020393 (Ack:64108606)] (S) Msg TX from 000000000001B669 to 1:0000000012344321 [23BF] [UDP:[fe80::16fb:ddf1:36af:fdc5%wlp0s20f3]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
> [MatterTest] 04-28 18:37:06.391 INFO 		***** Test Failure : 
> [MatterTest] 04-28 18:37:06.391 INFO Finished test in 695546.0ms
> [MatterTest] 04-28 18:37:06.391 ERROR 
> 
> ******************************************************************
> *
> * Test test_TC_AVSUM_2_7 failed for the following reason:
> * Details=<Status.NotFound: 139> != <Status.Success: 0>, Extras=None

**Command used to run the Python script:** python3 TC_AVSUM_2_7.py --commissioning-method on-network --discriminator 3840 --endpoint 1 --passcode 20202021   --storage-path admin_storage.json 

**Sample App used to run the script:** All-cluster-app 

### Bug prevalence

Whenever I'm executing this TC

### GitHub hash of the SDK that was being used

c3b6b762c843e6a487dda3d6636daf8a98adfe63

### Platform

raspi

### Anything else?

**Reference logs:**
[TC-AVSUM-2_7.txt](https://github.com/user-attachments/files/19940675/TC-AVSUM-2_7.txt)

**Python script References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_AVSUM_2_7.py

**Test plan References:**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/camera_av_settings_user_level_management.adoc#317-tc-avsum-27-dptzsetviewport-command-validation

marktrayer, 2025-06-28:
AVSUM 2_7 passes cleanly in the current camera app.
Could you please retest @Ashwinigrl

Ashwinigrl, 2025-07-15:
Issue resolved .

closed 2025-07-15 by Ashwinigrl