project-chip/connectedhomeip #38655: [CERT-TEST-FAILURE] [TC-LUNIT-3.1] Test step 5 failing due to TemperatureUnit attribute Status value mismatch.

Ashwinigrl, 2025-04-29:
### Feature Area

Other

### Test Case

TC-LUNIT-3.1

### Reproduction steps

1. Execute the below mentioned command to put DUT into a commissionable state :
` ./chip-all-clusters-app`

2. Once DUT reach the commissionable state sent below mentioned command on TH.
`./chip-tool pairing onnetwork 1 20202021`

**Once  successful pairing is done provide the below commands:**

1. TH reads from the DUT the LUNIT.S.A0001(SupportedTemperatureUnits) attribute

`./chip-tool unitlocalization read supported-temperature-units 1 0`

> [1745908005.827] [8463:8469] [TOO] Endpoint: 0 Cluster: 0x0000_002D Attribute 0x0000_0001 DataVersion: 3080355275
> [1745908005.827] [8463:8469] [TOO]   SupportedTemperatureUnits: 2 entries
> [1745908005.827] [8463:8469] [TOO]     [1]: 0
> [1745908005.827] [8463:8469] [TOO]     [2]: 1

2.  Construct a list with valid TempUnitEnum values that are not contained in SupportedUnitsList. Store the value as UnsupportedUnitsList.
   
`The value **2** is a valid TempUnitEnum value and not listed in the SupportedUnitList. Store this value as UnsupportedUnitsList.`

3. TH writes to the DUT with the UnsupportedUnitsList value on TemperatureUnit attribute.

 `./chip-tool unitlocalization write temperature-unit 2 1 0`

**Actual Outcome:**

**DUT should respond with CONSTRAINT_ERROR response instead of SUCCESS.**

> [1745908590.283] [8685:8691] [DMG]                         StatusIB =
> [1745908590.283] [8685:8691] [DMG]                         {
> **[1745908590.283] [8685:8691] [DMG]                                 status = 0x00 (SUCCESS),**
> [1745908590.283] [8685:8691] [DMG]                         },

**Expected Outcome:**

> [1745909595.616] [8977:8983] [DMG] 			StatusIB =
> [1745909595.616] [8977:8983] [DMG] 			{
> **[1745909595.617] [8977:8983] [DMG] 				status = 0x87 (CONSTRAINT_ERROR),**
> [1745909595.617] [8977:8983] [DMG] 			},

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

7460361eda292374a0a84f7dcc83f95047fc29e0

### Platform

raspi

### Anything else?

As per the Specification and Test Plan for the **UnitLocalization Cluster**, when a write command is sent to the **TemperatureUni**t attribute with a valid **TempUnitEnum** value that is not present in the **SupportedUnitsList** attribute, the DUT (All-cluster-app) should respond with a CONSTRAINT_ERROR (0x87). However, the current Sample app (All-cluster-app) is responding with SUCCESS. Therefore, the All-cluster app needs to be updated to return a CONSTRAINT_ERROR (0x87) instead of SUCCESS.

**Test step that require review:**

Test plan step 5: With each entry in UnsupportedUnitsList, TH writes to the DUT the TemperatureUnit attribute.

**Sample app used to run the Test:** All-cluster-app

**Specification References:**
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/LocalizationUnit.adoc#6-attributes

[image]

**Test plan References:**
-  https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/unitlocalization.adoc

[image]

**Reference logs:**

[TC-LNIT-3.1.txt](https://github.com/user-attachments/files/19954518/TC-LNIT-3.1.txt)

cecille, 2025-05-08:
@Ashwinigrl - are you running this test manually or using the new script?

Ashwinigrl, 2025-05-08:
@cecille I have run the test case manually.

Ashwinigrl, 2025-05-22:
Issue fixed in Commit f27301336812f41903a0ab97d77571c119320059. Hence closing it.

closed 2025-05-22 by Ashwinigrl