project-chip/connectedhomeip #38663: [BUG] [TC-ACL-2.6] Step 4 and 6 needs to be modified by updating the Privilege field and AuthMode field as per the test plan

KishokG, 2025-04-29:
### Reproduction steps

**The python script needs to be updated for TC-ACL-2.6 as per the test plan by modifying the Privilege field and AuthMode field:**

### Step-4: 

In step 4, the Privilege field and AuthMode field needs to be updated.

**Test Plan reference:** 

[image]

**Current Script reference:** 

[image]

### Step-6: 

In step 6, the Privilege field needs to be updated.

**Test Plan reference:** 

[image]

**Current Script reference:** 

[image]

### Bug prevalence

When

### GitHub hash of the SDK that was being used

s

### Platform

raspi

### Platform Version(s)

s

j-ororke, 2025-05-08:
Provided fix for the above issue can be found in PR #[38705](https://github.com/project-chip/connectedhomeip/pull/38705)

Alami-Amine, 2025-05-20:
- fixed by @j-ororke  in https://github.com/project-chip/connectedhomeip/pull/38705

closed 2025-05-20 by Alami-Amine