project-chip/connectedhomeip #38746: Segmentation fault during BLE unsubscribe after successful commissioning complete.

Saravana-kr22, 2025-05-05:
**Description: **

When executing the script TC_Pair.py as a part of stress testing, After successful commissioning over BLE-Thread, the controller encounters a segmentation fault while closing the BLE connection. The `ERROR C2 is NULL in UnsubscribeCharacteristicImpl` is logged just before the crash.

> [MatterTest/ble-thread] 05-01 01:47:14.115 INFO Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0005
[MatterTest/ble-thread] 05-01 01:47:14.115 INFO Received CommissioningComplete response, errorCode=0
[MatterTest/ble-thread] 05-01 01:47:14.116 INFO {
[MatterTest/ble-thread] 05-01 01:47:14.116 INFO 	"label" : "core_commissioning_stage_send_complete",
[MatterTest/ble-thread] 05-01 01:47:14.116 INFO 	"value" : 0
[MatterTest/ble-thread] 05-01 01:47:14.116 INFO }
[MatterTest/ble-thread] 05-01 01:47:14.116 INFO Successfully finished commissioning step 'SendComplete'
[MatterTest/ble-thread] 05-01 01:47:14.117 INFO Commissioning stage next step: 'SendComplete' -> 'Cleanup'
[MatterTest/ble-thread] 05-01 01:47:14.117 INFO {
[MatterTest/ble-thread] 05-01 01:47:14.117 INFO 	"label" : "core_dcm_commission_stage",
[MatterTest/ble-thread] 05-01 01:47:14.117 INFO 	"value" : 39
[MatterTest/ble-thread] 05-01 01:47:14.118 INFO }
[MatterTest/ble-thread] 05-01 01:47:14.118 INFO {
[MatterTest/ble-thread] 05-01 01:47:14.118 INFO 	"label" : "core_commissioning_stage_cleanup",
[MatterTest/ble-thread] 05-01 01:47:14.118 INFO 	"value" : null
[MatterTest/ble-thread] 05-01 01:47:14.118 INFO }
[MatterTest/ble-thread] 05-01 01:47:14.119 INFO Performing next commissioning step 'Cleanup'
[MatterTest/ble-thread] 05-01 01:47:14.119 INFO {
[MatterTest/ble-thread] 05-01 01:47:14.119 INFO 	"label" : "core_commissioning_stage_cleanup",
[MatterTest/ble-thread] 05-01 01:47:14.119 INFO 	"value" : 0
[MatterTest/ble-thread] 05-01 01:47:14.119 INFO }
[MatterTest/ble-thread] 05-01 01:47:14.120 INFO Successfully finished commissioning step 'Cleanup'
[MatterTest/ble-thread] 05-01 01:47:14.120 INFO Closing all BLE connections
[MatterTest/ble-thread] 05-01 01:47:14.120 ERROR C2 is NULL in UnsubscribeCharacteristicImpl
[MatterTest/ble-thread] 05-01 01:47:14.121 INFO Auto-closing end point's BLE connection.
[MatterTest/ble-thread] 05-01 01:47:14.121 INFO Closing BLE GATT connection (con 0x754950029030)
/home/grl/chip_jenkins_workspace/workspace/stress_test_pipeline/60/copied_controller_binaries@tmp/durable-78c9c596/script.sh.copy: line 5: 1708018 Segmentation fault      (core dumped)

GitHub hash of the SDK that were being used are : 09fbc4656cc3880be9b2c69168a9d5776b86a34f

**Test Bed :**
Controller Platform - python controller running on linux machine(amd64) ,
Nordic dk as DUT - all-cluster-app running on Nordic thread device,
Provisioning method - Ble-thread

**Test scenario**
Pair/Provision the DUT (All-cluster-app running on Nordic) using ble-thread.
Read device information after the device is provisioned
Unpair the DUT (i.e. removing the fabric)
Factory reset the DUT
repeat step1 - 4 for 100 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Execution Details:**

In this test-bed 48 iteration were passed and 49 iteration is crashed with this issue:

Passed iteartion log:  [controller_log_iteration_48_2025-05-01_01-46-34.log](https://github.com/user-attachments/files/20034087/controller_log_iteration_48_2025-05-01_01-46-34.log)
Crashed iteration log: [controller_log_iteration_49_2025-05-01_01-47-01.log](https://github.com/user-attachments/files/20034094/controller_log_iteration_49_2025-05-01_01-47-01.log)

Saravana-kr22, 2025-05-05:
@feasel0  Could you please take a look at this? The segmentation fault occurred once during the Friday run, but did not occur in the Monday execution.

feasel0, 2025-05-12:
In the failed iteration I see that `BLE connection closed` is appearing before `Closing BLE GATT connection`,

[image]

whereas in the successful iteration it happens after.

[image]

That could explain why [connection->mC2.get()](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/bluez/BluezConnection.cpp#L380) is coming back null.  That would prevent `bluez_gatt_characteristic1_call_stop_notify(pC2, ...)` from being called, which could potentially be leading to the segfault?

Is the failed iteration the only iteration in which `BLE connection closed` is happening before?  Or did any of the successful ones do this?

feasel0, 2025-05-12:
Two other things I noticed in the failed iteration log that may or may not be significant:

There is a debug warning `Device 37:D1:C3:17:13:82 does not look like a CHIP device.`

[image]

And there are some extra `StorageAdapter::GetKeyValue` and `StorageAdapter::DeleteKeyValue` operations that aren't happening in the successful iteration.  Two of which are showing `Buf not big enough` errors.

[image]

I'm not sure if this happens in any of the successful iterations.

closed 2025-07-16 by mergify[bot]