project-chip/connectedhomeip #38748: Commissioning Failure at WiFiNetworkEnable Step – Internal Error (CHIP Error 0x000000AC)

Saravana-kr22, 2025-05-05:
**Description:** 

When executing the script TC_Pair.py as a part of stress testing, the commissioning process fails at the WiFiNetworkEnable step with an internal error. The following error is observed in the logs:

> 2025-04-29 21:27:44,932 - chip.native.CTL - INFO - __init__.py:_RedirectToPythonLogging:41 - Commissioning complete for node ID 0x00000000D3BF130E: src/controller/CHIPDeviceController.cpp:2929: CHIP Error 0x000000AC: Internal error
2025-04-29 21:27:44,933 - chip.ChipDeviceCtrl - WARNING - ChipDeviceCtrl.py:HandleCommissioningComplete:474 - Failed to commission: src/controller/CHIPDeviceController.cpp:2929: CHIP Error 0x000000AC: Internal error

This issue occurred during commissioning specifically during iteration 13. The commissioning proceeds through initial steps, including receiving the ConnectNetwork response (networkingStatus=12), but fails when attempting to enable WiFi on the device. 

GitHub hash of the SDK that were being used are : cd3f2aa5d83fb7bfa23b8009f63d4d7ed3998b0d

**Test Bed :**
Controller Platform - python controller running on Raspberrypi 4B -8GB.
Raspberrypi as DUT - all-cluster-app running on Raspberrypi 4B -8GB.
Provisioning method - Ble-wifi

**Test scenario:**
Pair/Provision the DUT (All-cluster-app running on RPI) using ble-wifi.
Read device information after the device is provisioned
Unpair the DUT (i.e. removing the fabric)
Factory reset the DUT
repeat step1 - 4 for 100 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Pair.py

**Execution Details:**
In this Test bed  Iteration[13, 32, 33, 51, 52] are failed with incorect stare error below are the log:

Passed log of iteration 12:  [passed-12.zip](https://github.com/user-attachments/files/20034964/passed-12.zip)
Failed log of iteartion 13: [failed-13.zip](https://github.com/user-attachments/files/20034967/failed-13.zip)
Passed log of iteration 14: [passed-14.zip](https://github.com/user-attachments/files/20034972/passed-14.zip)

bzbarsky-apple, 2025-05-05:
> The commissioning proceeds through initial steps, including receiving the ConnectNetwork response (networkingStatus=12),

Status 12 is UnknownError.  As in, ConnectNetwork failed.

@Saravana-kr22 unfortunately the DUT log just shows ConnnectNetwork being received and then is cut off; presumably whatever log capture method is involved does not turn off buffering and then kills the relevant processes at some point before everything is flushed to disk.

So hard to tell _why_ the DUT failed ConnectNetwork.

enkiusz, 2025-05-09:
What is the Linux OS version on both of the raspberrys?

enkiusz, 2025-05-09:
Also does the controller raspberry pi serve as the WiFi AP or is there an external AP?

Harshith-GRL, 2025-05-12:
@enkiusz it is using Ubuntu os 24.04  as OS

bzbarsky-apple, 2025-05-20:
The SDK does not buffer output, but depending on how you are putting that output into a file (if you are) the shell could be buffering.

cjandhyala, 2025-06-05:
@Saravana-kr22 Could you collect dbus logs from controller and attach the logs here

KishokG, 2025-10-06:
@feasel0 Here are the latest log files for your reference:

[Latest_PASS_Failure_logs.zip](https://github.com/user-attachments/files/22721600/Latest_PASS_Failure_logs.zip)

bzbarsky-apple, 2025-10-06:
@enkiusz At least for non-concurrent-mode commissionees, the retries, if any, would have to be on the commissionee, since ConnectNetwork tears down the PASE session in those cases, right?