project-chip/connectedhomeip #38825: [darwin] Flaky test on Darwin framework tests - asan

andy31415, 2025-05-09:
See https://github.com/project-chip/connectedhomeip/actions/runs/14918836865/job/41910740444

```
2025-05-09T01:37:28.3414350Z Failing tests:
2025-05-09T01:37:28.3414780Z 
2025-05-09T01:37:28.3429690Z Test session results, code coverage, and logs:
2025-05-09T01:37:28.3433230Z 	/tmp/darwin/framework-tests/TestResults.xcresult
2025-05-09T01:37:28.3435450Z 
2025-05-09T01:37:28.3441460Z 	-[MTRPerControllerStorageTests testSubscriptionPool]
2025-05-09T01:37:28.3445330Z 	-[MTRPerControllerStorageTests testSubscriptionPoolManyDevices]
```

It is unclear which one fails - is it subscription pool?

bzbarsky-apple, 2025-05-09:
The primary failure here is in testSubscriptionPool:

```
2025-05-09T01:35:57.6032960Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/MTRPerControllerStorageTests.m:2902: error: -[MTRPerControllerStorageTests testSubscriptionPool] : Asynchronous wait failed: Exceeded timeout of 3 seconds, with unfulfilled expectations: "Fake device deleted".
```

and in fact, the dealloc of the MTRDevice does not happen until after this:

```
2025-05-09T01:35:57.7019210Z 2025-05-09 01:35:57.581129+0000 xctest[45811:128193] [Default] MTRDevice dealloc: 0x615000601600
```

Need to figure out why it did not get deallocated....

andy31415, 2025-05-13:
https://github.com/project-chip/connectedhomeip/actions/runs/14997823743/job/42138264912 also says:

```
2025-05-13T14:25:47.8599670Z Test session results, code coverage, and logs:
2025-05-13T14:25:47.8600430Z 	/tmp/darwin/framework-tests/TestResults.xcresult
2025-05-13T14:25:47.8600930Z 
2025-05-13T14:25:47.8603010Z Failing tests:
2025-05-13T14:25:47.8652320Z 	-[MTRPerControllerStorageTests testSubscriptionPoolManyDevices]
```

andy31415, 2025-05-14:
also in https://github.com/project-chip/connectedhomeip/pull/38866#issuecomment-2880082456

I will create a PR to disable this test, this has high flake rates.

bzbarsky-apple, 2025-05-15:
Should have been fixed in https://github.com/project-chip/connectedhomeip/pull/38854

closed 2025-05-15 by bzbarsky-apple