project-chip/connectedhomeip #38876: [Trivial] ## Issue: Updated attribute values not reflected in CHIP REPL

aravindone, 2025-05-12:
### Type

Other

### Explanation

Hello @bzbarsky-apple and @jamesharrow,

I'm trying to update and verify attribute values using the `ElectricalPowerMeasurementDelegate`. I used the `Set` functions to assign values and then used the corresponding `Get` functions to read them back. However, while the values appear to be set correctly in logs, they are not reflected in the CHIP REPL output.

Here is the code snippet I’m using:

```
static ElectricalPowerMeasurementDelegate gPowerDelegate;
void SimulateMeasurements()
{
    ChipLogProgress(AppServer, "Inside simulate measurement");
    int64_t voltage = 230;  // Example: 230V
    int64_t power   = 50;   // Example: 50W
    int64_t current = 10;   // Example: 10A

    // Set attributes
    gPowerDelegate.SetVoltage(DataModel::Nullable<int64_t>(voltage));
    gPowerDelegate.SetActivePower(DataModel::Nullable<int64_t>(power));
    gPowerDelegate.SetActiveCurrent(DataModel::Nullable<int64_t>(current));

    // Get attributes back and print
    auto readVoltage = gPowerDelegate.GetVoltage();
    auto readPower   = gPowerDelegate.GetActivePower();
    auto readCurrent = gPowerDelegate.GetActiveCurrent();

    if (!readVoltage.IsNull())
    {
        ChipLogProgress(AppServer, "Voltage after setting: %lld V", readVoltage.Value());
    }
    else
    {
        ChipLogError(AppServer, "Voltage is NULL after setting!");
    }

    if (!readPower.IsNull())
    {
        ChipLogProgress(AppServer, "Power after setting: %lld W", readPower.Value());
    }
    else
    {
        ChipLogError(AppServer, "Power is NULL after setting!");
    }

    if (!readCurrent.IsNull())
    {
        ChipLogProgress(AppServer, "Current after setting: %lld A", readCurrent.Value());
    }
    else
    {
        ChipLogError(AppServer, "Current is NULL after setting!");
    }
}
```

output image:
[image]

Could you please guide me on what might be missing or if there’s an additional step required to sync these values with the actual Matter attribute storage?

Looking forward to your feedback.

Thanks!

### Platform

esp32

### Platform Version(s)

1.3

### Testing

Platform validated

bzbarsky-apple, 2025-05-13:
@aravindone There is no "actual Matter attribute storage".  When read, Matter just gets the values from the delegate set up on the cluster instance.

jamesharrow, 2025-05-13:
I suggest you take a look at the code here:

https://github.com/project-chip/connectedhomeip/blob/master/examples/energy-management-app/energy-management-common/energy-reporting/src/FakeReadings.cpp

and how the EVSE delegate has a function call SendPowerReadings() 

https://github.com/project-chip/connectedhomeip/blob/fdacd16c5b6d84aac37ed15cd411312430bb095c/examples/energy-management-app/energy-management-common/energy-evse/src/EVSEManufacturerImpl.cpp#L390

Also NOTE: all values are in mV, mW, mA (not volts, watts, amps) - so you need to multiply all of your values by 1000 !

aravindone, 2025-05-15:
Thanks both im closing this ticket now

closed 2025-05-15 by aravindone