project-chip/connectedhomeip #38894: Fix AdminCommissioning cluster spec-sdk mismatch.

tersal, 2025-05-12:
It seems that there is a minor change in the StatusCode enum from AdministratorCommissioning.

The generated one based on the specs is called StatusCodeEnum instead of plainly StatusCode.

Old:
`// Enum for StatusCode
enum class StatusCode : uint8_t
{
    kBusy               = 0x02,
    kPAKEParameterError = 0x03,
    kWindowNotOpen      = 0x04,
    // All received enum values that are not listed above will be mapped
    // to kUnknownEnumValue. This is a helper enum value that should only
    // be used by code to process how it handles receiving and unknown
    // enum value. This specific should never be transmitted.
    kUnknownEnumValue = 0,
};`

New:
`// Enum for StatusCodeEnum
enum class StatusCodeEnum : uint8_t
{
    kBusy               = 0x02,
    kPAKEParameterError = 0x03,
    kWindowNotOpen      = 0x04,
    // All received enum values that are not listed above will be mapped
    // to kUnknownEnumValue. This is a helper enum value that should only
    // be used by code to process how it handles receiving and unknown
    // enum value. This specific should never be transmitted.
    kUnknownEnumValue = 0,
};`

It actually looks like a minor change but

bzbarsky-apple, 2025-05-13:
We should be able to fix the name as long as we add the right bits to Darwin availability renaming stuff and `src/app/common/CompatEnumNames.h`, right?

bzbarsky-apple, 2025-07-03:
Yes, but there is also a lot of value in having SDK APIs match the spec when we can manage it...