project-chip/connectedhomeip #38897: Fix ColorControl cluster mismatch

tersal, 2025-05-13:
There are some weird differences between the Alchemy generated (based on the spec) and the existing SDK implementation for color control.

The most meaningful one seems to be the change from `uint8_t` to `ColorLoopDirectionEnum` for the `ColorLoopDirection` attribute.

Before:
```c
namespace ColorLoopDirection {
struct TypeInfo
{
    using Type             = uint8_t;
    using DecodableType    = uint8_t;
    using DecodableArgType = uint8_t;

    static constexpr ClusterId GetClusterId() { return Clusters::ColorControl::Id; }
    static constexpr AttributeId GetAttributeId() { return Attributes::ColorLoopDirection::Id; }
    static constexpr bool MustUseTimedWrite() { return false; }
};
} // namespace ColorLoopDirection
```

After:
```c
namespace ColorLoopDirection {
struct TypeInfo
{
    using Type             = chip::app::Clusters::ColorControl::ColorLoopDirectionEnum;
    using DecodableType    = chip::app::Clusters::ColorControl::ColorLoopDirectionEnum;
    using DecodableArgType = chip::app::Clusters::ColorControl::ColorLoopDirectionEnum;

    static constexpr ClusterId GetClusterId() { return Clusters::ColorControl::Id; }
    static constexpr AttributeId GetAttributeId() { return Attributes::ColorLoopDirection::Id; }
    static constexpr bool MustUseTimedWrite() { return false; }
};
} // namespace ColorLoopDirection
```

Also the feature bitmap had a change in one of the enum elements generated from `kHueAndSaturation` to `kHueSaturation`.

Before:
```c
// Bitmap for Feature
enum class Feature : uint32_t
{
    kHueAndSaturation = 0x1,
    kEnhancedHue      = 0x2,
    kColorLoop        = 0x4,
    kXy               = 0x8,
    kColorTemperature = 0x10,
};
```

After:
```c
// Bitmap for Feature
enum class Feature : uint32_t
{
    kHueSaturation    = 0x1,
    kEnhancedHue      = 0x2,
    kColorLoop        = 0x4,
    kXy               = 0x8,
    kColorTemperature = 0x10,
};
```

bzbarsky-apple, 2025-05-13:
@tersal Please note https://github.com/project-chip/connectedhomeip/issues/30359

pimpalemahesh, 2026-02-26:
@tersal @bzbarsky-apple Do you have any updates on this?

Also, regarding this comment:
https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/12526#issuecomment-3559788170

Does this mean that it cannot be updated?