project-chip/connectedhomeip #38900: Error during ZAP and Alchemy generation for DoorLock cluster

tersal, 2025-05-13:
The `DoorLock` cluster seems to have a couple of issues when trying to generate the code with ZAP based on the Alchemy generated XML.

1.- There is an error during the ZAP generation
```
lark.exceptions.UnexpectedToken: Unexpected token Token('ID', 'desc') at line 5803, column 3.
Expected one of: 
	* COMMAND
	* SHARED
	* PROVISIONAL
	* TIMEDWRITE
	* RBRACE
	* READONLY
	* INTERNAL
	* BITMAP
	* CRITICAL
	* DEBUG
	* ATTRIBUTE
	* INFO
	* STABLE
	* FABRIC_SCOPED
	* DEPRECATED
	* NOSUBSCRIBE
	* STRUCT
	* REQUEST
	* TIMED
	* ENUM
	* FABRIC
	* FABRIC_SENSITIVE
	* RESPONSE
Previous tokens: [Token('RBRACE', '}')]
```

2.- The `LEDSettings` attribute type changed from `uint8_t` to `LEDSettingEnum`, this may break compatibility.

Before:
```c
namespace LEDSettings {
struct TypeInfo
{
    using Type             = uint8_t;
    using DecodableType    = uint8_t;
    using DecodableArgType = uint8_t;

    static constexpr ClusterId GetClusterId() { return Clusters::DoorLock::Id; }
    static constexpr AttributeId GetAttributeId() { return Attributes::LEDSettings::Id; }
    static constexpr bool MustUseTimedWrite() { return false; }
};
} // namespace LEDSettings
```

After:
```c
namespace LEDSettings {
struct TypeInfo
{
    using Type             = chip::app::Clusters::DoorLock::LEDSettingEnum;
    using DecodableType    = chip::app::Clusters::DoorLock::LEDSettingEnum;
    using DecodableArgType = chip::app::Clusters::DoorLock::LEDSettingEnum;

    static constexpr ClusterId GetClusterId() { return Clusters::DoorLock::Id; }
    static constexpr AttributeId GetAttributeId() { return Attributes::LEDSettings::Id; }
    static constexpr bool MustUseTimedWrite() { return false; }
};
} // namespace LEDSettings
```

bzbarsky-apple, 2025-05-13:
@tersal note https://github.com/project-chip/connectedhomeip/issues/30303

But also, it's hard to say anything about the generation error without knowing what the Alchemy-generated XML looks like.  It sure looks to me like the actual error is during parsing of .matter files, and this might happen if you use an old parser with the current tip, since @andy31415 changed the syntax recently.

andy31415, 2025-06-09:
This is using an old matter_idl parser. need to update to master and fix things.

closed 2025-06-09 by andy31415