project-chip/connectedhomeip #38908: [SVE] IDM_10.4 asserted due to self.endpointid

jamesharrow, 2025-05-13:
### Reproduction steps

1. Running DUT against test harness I selected a few IDM test cases to run including IDM 10.4
NOTE in my config I did not include any endpoint ID setting.

2. when it ran it looks like it hit an assert due to missing endpoint_id?

[MatterTest] 05-13 13:54:31.994 ERROR Exception occurred in test_TC_IDM_10_4.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/dist-packages/mobly/base_test.py", line 818, in exec_one_test
    test_method()
  File "/root/python_testing/scripts/sdk/TC_pics_checker.py", line 99, in test_TC_IDM_10_4
    self.endpoint = self.endpoints_tlv[self.endpoint_id]
                    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
KeyError: None

### Bug prevalence

Tested once with test harness

### GitHub hash of the SDK that was being used

f4d171ad4e043c1d5b53481660a618b60d8a1222

### Platform

python

### Type

Test Improvement

### Testing

Manually tested with SDK

### (Optional) If manually tested please explain why this is only manually tested

There's no option on the 'New Issue' to select using the Matter Test Harness

### Anything else?

May be user error on my part - its not clear if endpoints need to be set in the test harness somehow?

jamesharrow, 2025-05-13:
[TC_IDM_10_4_Logs.log](https://github.com/user-attachments/files/20191546/TC_IDM_10_4_Logs.log)

tcarmelveilleux, 2025-05-16:
This test must always be run with an endpoint since PICS lists don't yet support cleanly expressing PICS for all endpoints at once. Can you please update the test to immediately fail at the start if `--endpoint` is not provided?

jamesharrow, 2025-05-17:
Looks like @cecille had added a fix already in#39042

jamesharrow, 2025-05-17:
We may also want an issue on the test harness to make it automatic on which EP to set up (seems that now PICS tool is able to specify the endpoint then it shouldn’t need the test config to be manually set for each test).

jamesharrow, 2025-06-06:
This has merged so closing out this issue: https://github.com/project-chip/connectedhomeip/pull/39042

closed 2025-06-06 by jamesharrow