project-chip/connectedhomeip #38933: [BUG] Build issue in Camera-App (RPi 5)

joyoungjun, 2025-05-14:
### Reproduction steps

Command  used to build the camera-app(Sample App)
 ./scripts/examples/gn_build_example.sh examples/camera-app/linux/ examples/camera-app/linux/out/camera-app chip_inet_config_enable_ipv4=false

Logs
`2025-05-14 01:47:11,163 Loading extra packages for linux
2025-05-14 01:47:11,163 Skipping: darwin (i.e. /home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json)
2025-05-14 01:47:11,163 Skipping: windows (i.e. /home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json)

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █
 ▀▀    █   █    ▀▀

  ACTIVATOR! This sets your shell environment variables.

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20250514 01:47:11 INF Environment passes all checks!

Environment looks good, you are ready to go!

+ env
SHELL=/bin/bash
PW_PROJECT_ROOT=/home/ubuntu/connectedhomeip
PW_ZAP_CIPD_INSTALL_DIR=/home/ubuntu/connectedhomeip/.environment/cipd/packages/zap
PWD=/home/ubuntu/connectedhomeip
LOGNAME=ubuntu
XDG_SESSION_TYPE=tty
CIPD_CACHE_DIR=/home/ubuntu/.cipd-cache-dir
PW_ROOT=/home/ubuntu/connectedhomeip/third_party/pigweed/repo
HOME=/home/ubuntu
LANG=C.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.crdownload=00;90:*.dpkg-dist=00;90:*.dpkg-new=00;90:*.dpkg-old=00;90:*.dpkg-tmp=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:*.swp=00;90:*.tmp=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:
VIRTUAL_ENV=/home/ubuntu/connectedhomeip/.environment/pigweed-venv
LC_TERMINAL=iTerm2
SSH_CONNECTION=192.168.1.31 52636 192.168.1.32 22
_PW_ROSETTA=0
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=ubuntu
LC_TERMINAL_VERSION=3.5.13
SHLVL=2
PW_PACKAGE_ROOT=/home/ubuntu/connectedhomeip/.environment/packages
XDG_SESSION_ID=16
_PW_ENVIRONMENT_CONFIG_FILE=/home/ubuntu/connectedhomeip/scripts/setup/environment.json
XDG_RUNTIME_DIR=/run/user/1000
PW_BRANDING_BANNER=/home/ubuntu/connectedhomeip/scripts/setup/banner.txt
_PW_ACTUAL_ENVIRONMENT_ROOT=/home/ubuntu/connectedhomeip/.environment
SSH_CLIENT=192.168.1.31 52636 22
PW_ARM_CIPD_INSTALL_DIR=/home/ubuntu/connectedhomeip/.environment/cipd/packages/arm
XDG_DATA_DIRS=/usr/local/share:/usr/share:/var/lib/snapd/desktop
PATH=/home/ubuntu/connectedhomeip/third_party/pigweed/repo/out/host/host_tools:/home/ubuntu/connectedhomeip/.environment/pigweed-venv/bin:/home/ubuntu/connectedhomeip/.environment/cipd/packages/arm/bin:/home/ubuntu/connectedhomeip/.environment/cipd/packages/arm:/home/ubuntu/connectedhomeip/.environment/cipd/packages/zap:/home/ubuntu/connectedhomeip/.environment/cipd/packages/pigweed/bin:/home/ubuntu/connectedhomeip/.environment/cipd/packages/pigweed:/home/ubuntu/connectedhomeip/.environment/cipd:/home/ubuntu/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
[... 418 lines cut ...]
      |                                           ^~~~~~~~
/usr/include/gstreamer-1.0/gst/gstutils.h:849:3: note: in expansion of macro ‘GST_WRITE_UINT64_BE’
  849 |   GST_WRITE_UINT64_BE (data, u.i);
      |   ^~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
ubuntu@ubuntu:~/connectedhomeip$`

### Bug prevalence

whenever I do this

### GitHub hash of the SDK that was being used

b8b0116b2c71e1db9e3db23d5596083da1badb51

### Platform

raspi

### Platform Version(s)

Raspberrypi 5

kiel-apple, 2025-05-14:
fingers crossed this is solved by #38770

joyoungjun, 2025-05-15:
@kiel-apple Thanks for letting me know, but unfortunately it doesn't solve the problem. I'm still getting the same error.

kiel-apple, 2025-05-15:
because the errors are in `gstreamer`, I think the issue is with the the distribution-provided `gstreamer` package rather than with code on the Matter side, which will make it more challenging to resolve.

joyoungjun, 2025-05-15:
Oh, then it will take some time to resolve. Thanks for letting me know.

closed 2025-09-10 by joyoungjun