project-chip/connectedhomeip #38935: [BUG] [TC-ACT-2.2 & 3.2] Getting UNSUPPORTED_CLUSTER response while sending Instant Action command from the Actions cluster in Bridge-app

KishokG, 2025-05-14:
### Reproduction steps

1. Bring up the DUT with Bridge-app.
2. Pair the DUT using chip-tool through "Onnetwork" transport.
3. Run the command `./chip-tool actions read accepted-command-list 1 1`. From the DUT response you can see the InstantAction command is implemented in the Bridge-app.

**DUT Response:** 

> [1747204998.043] [17492:17498] [TOO] Endpoint: 1 Cluster: 0x0000_0025 Attribute 0x0000_FFF9 DataVersion: 3592451377
> [1747204998.043] [17492:17498] [TOO]   AcceptedCommandList: 1 entries
> [1747204998.043] [17492:17498] [TOO]     [1]: 0 (InstantAction)

4. Run the command: `./chip-tool actions  instant-action  0x1001 1 1`

**Expected outcome:** 

> [1659955888.114483][23216:23221] CHIP:DMG:                                 StatusIB =
> [1659955888.114499][23216:23221] CHIP:DMG:                                 {
> [1659955888.114515][23216:23221] CHIP:DMG:                                         status = 0x00 (SUCCESS),
> [1659955888.114529][23216:23221] CHIP:DMG:                                 },

**Current DUT response:** 

> [1747205164.948] [17510:17516] [DMG] 					EndpointId = 0x1,
> [1747205164.948] [17510:17516] [DMG] 					ClusterId = 0x25,
> [1747205164.948] [17510:17516] [DMG] 					CommandId = 0x0,
> [1747205164.948] [17510:17516] [DMG] 				},
> [1747205164.948] [17510:17516] [DMG] 				
> [1747205164.948] [17510:17516] [DMG] 				StatusIB =
> [1747205164.948] [17510:17516] [DMG] 				{
> [1747205164.948] [17510:17516] [DMG] 					status = 0xc3 (UNSUPPORTED_CLUSTER),
> [1747205164.948] [17510:17516] [DMG] 				},

### Bug prevalence

Whenever I run this command

### GitHub hash of the SDK that was being used

a12250a4e4ffce5be5f809479fbe8d85e5acd230

### Platform

raspi

### Anything else?

**Affected test cases and test steps:**

1. TC-ACT-2.2: Step 4d
2. TC-ACT-3.2: Step 1

bzbarsky-apple, 2025-05-15:
@andy31415 This is pretty weird... how are we managing to get a status that does not match the AcceptedCommandList?  Clearly the AcceptedCommandList is reading some other source of truth than whatever is doing command dispatch, which sounds like a bug in the provider bits?

andy31415, 2025-05-15:
Validated the report in repl:
  - cluster `Actions` exists on endpoint 1
  - cluster reports `0` as accepted commands list, which should be instantaction
  - Trying to send the command does indeed return unsupported cluster:

[image]

andy31415, 2025-05-15:
what seems to be happening is that the cluster is marked as `CommandHandlerInterfaceonlyClusters` in https://github.com/project-chip/connectedhomeip/blob/master/src/app/common/templates/config-data.yaml#L62

So as a result, the dispatch logic is not added in `DispatchCommand` inside ember and the result is "unsupported cluster" rather than invalid command as no CHI seems to be registered on this cluster.

andy31415, 2025-05-15:
It seems that the action sserver is expected to be initialized. THis is done in telink/esp32/asr but not in linux. As a result CHI never works.

andy31415, 2025-05-15:
Looking at code, `bridged-actions-stub` seems to initialize things. However compared to other implementations, no delegate seems to be used and direct registration is done.

andy31415, 2025-05-15:
Issue is that https://github.com/project-chip/connectedhomeip/blob/master/examples/bridge-app/linux/bridged-actions-stub.cpp#L134 initializes the attribute access but NOT the command interfaces, so things do not work as expected (no command support).

closed 2025-09-19 by KishokG