project-chip/connectedhomeip #38941: pairing Duration Increases Gradually in Multi-Admin Commissioning Stress Test

Saravana-kr22, 2025-05-14:
**Description:**

While running a multi-admin stress test, We observe progressively increasing latency in all the operations across iterations.

Pairing duration was keep on increasing gardually in 1st iteration total pairing duration for 15 controller is 33 sec but in the 38th iteration it is 272 and it keep on increasing 

[image]

In the same way the time duration taken for reading the subcription report from the dut and unpairing the dut is also increasing progrerssively after each iterations.

[image]

[image]

GitHub hash of the SDK that were being used are : 4e38905c6688effc074abecb34194a823ffbe5e5

**Test Bed :**
Controller Platform - python controller running on linux machine x86-64.
Raspberrypi as DUT - all-cluster-app running on Raspberrypi 4B -8GB.
Provisioning method - On-network

**Test scenario:**

Pair/Provision the DUT (All-cluster-app running on RPI) using on-network.
Read the SupportedFabrics attribute
Commission the node to the maximum number of controllers (SupportedFabrics).
Subscribe to the NodeLabel attribute from each controller.
Modify the NodeLabel attribute and ensure all controllers receive the update.
Read device information after the device is provisioned
Unpair all controllers one by one.
repeat step3 - 7 for 100 iterations

Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Multiadmin.py

**logs:**

iteration-1: [1.zip](https://github.com/user-attachments/files/20204774/1.zip)
iteration-10:[10.zip](https://github.com/user-attachments/files/20204791/10.zip)
iteration-20: [20.zip](https://github.com/user-attachments/files/20204795/20.zip)
iteration-35: [35.zip](https://github.com/user-attachments/files/20204799/35.zip)

bzbarsky-apple, 2025-05-15:
Probably the toy Linux platform storage is at fault here.

Saravana-kr22, 2025-05-16:
@bzbarsky-apple We are facing this is issue only on the DUT running on linux-RPI . So is there any method or API availble to clear this cache?

bzbarsky-apple, 2025-05-20:
Well, you should start by profiling to see whether that's the issue...

Saravana-kr22, 2025-07-04:
Hi @bzbarsky-apple , 

The profiling data is quite large, and unfortunately, we’re not able to open it directly on the RPI due to resource constraints. I tried extracting it using the command below:

`perf script > perf_report.txt`

This generated a text file with the profiling data, but it ended up consuming around 40 GB of disk space.

If you have the resources available, would you mind trying to generate it on your side using the same command and checking whether it works for you?

bzbarsky-apple, 2025-07-17:
@Saravana-kr22 I do not have a setup for reproducing this (e.g. no Linux setup on my end) or time to work on this.

This issue could benefit from clear steps to reproduce (i.e. the _exact_ commands to run after checking out the repository, in order) to reproduce the problem, and then needs someone to try reproducing this while profiling.

Saravana-kr22, 2025-07-24:
To reproduce the issue, you'll need:

- A Raspberry Pi (RPI) running the all-clusters-app as the DUT.
- A controller machine with:

         -Python chip controller environment.
         -Cloned and configured Matter QA repository.

**Steps to reproduce:**

1. Clone the Matter QA repo on the controller machine: https://github.com/CHIP-Specifications/matter-qa.
2. Build & activate the Matter Python environment as per:  https://github.com/project-chip/connectedhomeip/blob/master/docs/development_controllers/chip-repl/python_chip_controller_building.md
3. Install matter_qa package in the activated Python environment: ` pip install <path_to_matter_qa_repo>/matter_qa`
4. Update the config.yaml with your system details  as below

```
general_configs:
  platform_execution: "rpi" # Use 'rpi' option for using raspberrypi as DUT, for any other option used like 'nordic' or 'CustomDUT' same name will appear in log viewer UI and 'deviceModules' option from config will be read/used
  number_of_iterations: 3
  sleep_duration_between_iterations: 5  # Accept the int value for the sleep duration between the iterations if this parameter is missing a random value between 1- 5.
  delay_between_stress_test_operations: 2 # Provide delay between operations so that DUT can be put into a stable state

  analytics_parameters:
    current_heap_used :
      attribute: SoftwareDiagnostics.Attributes.CurrentHeapUsed
      end_point: 0
    reboot_count :
      attribute: GeneralDiagnostics.Attributes.RebootCount
      end_point: 0

dut_config:
  rpi: # These parameters are used to take ssh session to a remote host and start DUT application on a raspberrypi
    module: matter_qa.library.platform.raspberrypi # Python file which will inherit the BaseDUTNode class and define the behaviour of the DUT Working
    rpi_hostname: "rpi's ip address"
    delay_after_factory_reset: 5
    do_factory_reset_every_iteration: True
    rpi_username: "username of rpi"
    rpi_password: "password of rpi"
    rpi_sudo_password: "password of rpi"
    download_var_logs: False
    var_log_filenames: ["avahi-daemon.log"]
    app_config: # These parameters are used to start the DUT application os user's choice, provide full app path and the arguments
      matter_app: "/home/ubuntu/apps/chip-all-clusters-app"
      discriminator: 3000
      passcode: 20202021
test_case_config:
  TC_Pair:
    retry_count: 3
  TC_Multiadmin:
    open_commissioning_window_timeout: 300

```
5. Navigate to the  directory  <path/to/cloned/stress/test/code/repo>/matter_qa/src/scripts/reliability_scripts  Where the reliability scripts are located.
6. Execute the Multiadmin script with the command: `python3 TC_Multiadmin.py --discriminator 3790 --passcode 20202021 --storage-path admin_storage.json --timeout 9000000 --commissioning-method on-network --logs-path <path/to/store/execution/logs> --reliability-tests-arg <path/to/configFile.yaml> --trace-to json:log --paa-trust-store-path <path-to-connectedhomeip>/connectedhomeip/credentials/development/paa-root-certs/`

For more details, refer to the  https://github.com/CHIP-Specifications/matter-qa/blob/main/README.md 
If you have any questions or run into issues, feel free to reach out.

Saravana-kr22, 2025-08-01:
Hi @enkiusz , Thanks for the update.

Could you please confirm whether this is a platform-specific issue on Linux, or if it’s related to the SDK itself? Just trying to understand the scope of the problem better.

Saravana-kr22, 2025-08-11:
Hi @enkiusz ,

Thanks for your detailed investigation and the proof-of-concept fix.
I’ve gone ahead and raised an issue in the matter-qa repository to address this problem by avoiding unnecessary stdout/stderr buffering in pyinvoke.run() calls.

Saravana-kr22, 2025-09-03:
We updated the script to write the logs directly to the file instead of storing them in the buffer. As a result, the pairing duration has been reduced. Hence closing this issue, if required will raise another issue.

closed 2025-09-03 by Saravana-kr22