project-chip/connectedhomeip #38957: Missing attribute `DeviceLocation` ZCL data-model

lboue, 2025-05-14:
Missing attribute "DeviceLocation" in [zcl/data-model/chip/bridged-device-basic-information-cluster.xml](

https://github.com/project-chip/connectedhomeip/blob/162df095e28c60edff6bdbb6380a5b5e43c7e067/src/app/zap-templates/zcl/data-model/chip/bridged-device-basic-information-cluster.xml#L128-L131):

It should look like this:
```xml
    <attribute code="0x0017" side="server" name="DeviceLocation" define="DEVICE_LOCATION" type="LocationDescriptorStruct" default="MS" optional="true">
      <otherwiseConform>
        <provisionalConform/>
        <optionalConform/>
      </otherwiseConform>
    </attribute>
```

But exists here:
https://github.com/project-chip/connectedhomeip/blob/162df095e28c60edff6bdbb6380a5b5e43c7e067/data_model/master/clusters/BasicInformationCluster.xml#L354-L361

bzbarsky-apple, 2025-05-15:
Duplicate of #33555

closed 2025-05-15 by bzbarsky-apple

lboue, 2025-05-15:
Thanks. This ticket [#33555](https://github.com/project-chip/connectedhomeip/issues/33555) has been open for almost a year.