project-chip/connectedhomeip #38974: [BUG] [TC-SC-4.3] failed due to error 'NoneType' object has no attribute 'txt_record'

YuanWangSE, 2025-05-15:
### Reproduction steps

Execute the case in TH UI, then download the log.

*
* Test test_TC_SC_4_3 failed for the following reason:
* 'NoneType' object has no attribute 'txt_record'
*
* File "/root/python_testing/scripts/sdk/TC_SC_4_3.py", line 284, in test_TC_SC_4_3
* AttributeError: 'NoneType' object has no attribute 'txt_record'
*
* Test step:
*     7: TH performs a query for the TXT record against the qname instance_qname.	Expected outcome: Verify TXT record is returned
*
* Endpoint: None
*

### Bug prevalence

always

### GitHub hash of the SDK that was being used

a12250a4e4ffce5be5f809479fbe8d85e5acd230

### Platform

windows

pankore, 2025-06-04:
Hi, thank you for raising this issue.

I’m also encountering the same error when executing the UI Python test case in the TH using the default PICS files for the Raspberry Pi platform provided by QA in https://groups.csa-iot.org/wg/matter-csg/document/39613. The error persists when commissioning with the chip-all-cluster-app, and my test results align using my own DUT and TH having the same result with chip-all-cluster-app. For clarity and easier debugging, I’ve attached the relevant TH log and chip-all-cluster-app log during commissioning with the TH.
Thank you.

[TC-SC-4.3_allClusterApp.txt](https://github.com/user-attachments/files/20591169/TC-SC-4.3_allClusterApp.txt)

Test test_TC_SC_4_3 failed for the following reason:
Test Step 10 : TH performs a DNS-SD browse for _I<hhhh>._sub._matter._tcp.local, where <hhhh> is the 64-bit compressed Fabric identifier, expressed as a fixed-length, sixteencharacter hexadecimal string, encoded as ASCII (UTF-8) text using capital letters.
[MatterTest] 06-04 12:04:23.877 INFO MDNS discovered service types: ['_device-info._tcp.local.', '_matterd._udp.local.', '_smb._tcp.local.']
[MatterTest] 06-04 12:04:23.881 ERROR Exception occurred in test_TC_SC_4_3.

cecille, 2025-06-20:
https://github.com/project-chip/connectedhomeip/pull/39623 - fix linked here. Please review.

raul-marquez-csa, 2025-07-10:
Closed as fix has been merged

closed 2025-07-10 by raul-marquez-csa

Sandyyeow-realtek, 2025-07-22:
Hi, I attempted to retest the chip-all-cluster-app using the updated test script available at https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_SC_4_3.py. 
Unfortunately, the error persists. The test still fails during commissioning with the chip-all-cluster-app, and the test results are  consistent when using my own DUT along with the updated test script.

To assist investigation, I’ve attached the relevant logs:

[TC-SC-4.3.zip](https://github.com/user-attachments/files/21361128/TC-SC-4.3.zip)

Failure point: 
Test test_TC_SC_4_3 failed for at Test Step 10 : TH performs a DNS-SD browse for _I._sub._matter._tcp.local, where is the 64-bit compressed Fabric identifier, expressed as a fixed-length, sixteencharacter hexadecimal string, encoded as ASCII (UTF-8) text using capital letters.

Thank you.