project-chip/connectedhomeip #38978: [BUG] Exception thrown when sending MTRCommandPath over NSXPCConnection

Gernot, 2025-05-15:
### Reproduction steps

I created a MTRBaseDevice, and sent a command to my vacuum cleaner like this:
```swift
let controller = MTRDeviceController.sharedController(
            withID: home.matterControllerID as NSCopying,
            xpcConnect: home.matterControllerXPCConnectBlock
        )

//...

let device = MTRBaseDevice(nodeID: NSNumber(value: nodeID), controller: controller)
let vacuumCluster = MTRBaseClusterRVCOperationalState(
    device: device,
    endpointID: NSNumber(value: 1),
    queue: .main
)

let params = MTRRVCOperationalStateClusterResumeParams()
let result = try await vacuumCluster?.resume(
    with: params
)
```

This approach seems to work for different devices/cluster, but on my *Roborock QRevo Curv* vacuum cleaner, using Firmware *02.26.38* it fails with the following exception:

```
<NSXPCConnection: 0x132e76d00> connection to service with pid 224 named com.apple.matter.framework.xpc: Exception caught during decoding of reply to message 'invokeCommandWithController:nodeId:endpointId:clusterId:commandId:fields:timedInvokeTimeout:completion:', dropping incoming message and calling failure block.

Ignored Exception: Exception while decoding argument 0 (#1 of invocation):
<NSInvocation: 0x1330ecb80>
return value: {v} void
target: {@?} 0x0 (block)
argument 1: {@} 0x0
argument 2: {@} 0x0

Exception: value for key 'NS.objects' was of unexpected class 'MTRCommandPath' (0x28444d8b8) [/System/Library/Frameworks/Matter.framework].
Allowed classes are:
 {(
    "'NSError' (0x205f8f3e0) [/System/Library/Frameworks/Foundation.framework]",
    "'NSDate' (0x205f87508) [/System/Library/Frameworks/CoreFoundation.framework]",
    "'NSString' (0x205f8f458) [/System/Library/Frameworks/Foundation.framework]",
    "'NSArray' (0x205f879e0) [/System/Library/Frameworks/CoreFoundation.framework]",
    "'NSNumber' (0x205f8f480) [/System/Library/Frameworks/Foundation.framework]",
    "'NSData' (0x205f87760) [/System/Library/Frameworks/CoreFoundation.framework]",
    "'NSDictionary' (0x205f878a0) [/System/Library/Frameworks/CoreFoundation.framework]"
)}
(
	0   CoreFoundation                      0x000000019b34b228 7821F73C-378B-3A10-BE90-EF526B7DBA93 + 1155624
	1   libobjc.A.dylib                     0x00000001987e5abc objc_exception_throw + 88
	2   Foundation                          0x0000000199ec4a04 34DE055D-8683-380A-9198-C3347211D13D + 104964
	3   Foundation                          0x0000000199ec84ec 34DE055D-8683-380A-9198-C3347211D13D + 120044
	4   Foundation                          0x0000000199ec8ab8 34DE055D-8683-380A-9198-C3347211D13D + 121528
	5   Foundation                          0x0000000199ec8bec 34DE055D-8683-380A-9198-C3347211D13D + 121836
	6   Foundation                          0x0000000199ec3db8 34DE055D-8683-380A-9198-C3347211D13D + 101816
	7   Foundation                          0x0000000199ec6fb0 34DE055D-8683-380A-9198-C3347211D13D + 114608
	8   Foundation                          0x0000000199ec852c 34DE055D-8683-380A-9198-C3347211D13D + 120108
	9   Foundation                          0x0000000199ec8ab8 34DE055D-8683-380A-9198-C3347211D13D + 121528
	10  Foundation                          0x0000000199ec8bec 34DE055D-8683-380A-9198-C3347211D13D + 121836
	11  Foundation                          0x0000000199ec3db8 34DE055D-8683-380A-9198-C3347211D13D + 101816
	12  Foundation                          0x0000000199ed6154 34DE055D-8683-380A-9198-C3347211D13D + 176468
	13  Foundation                          0x0000000199ec852c 34DE055D-8683-380A-9198-C3347211D13D + 120108
	14  Foundation                          0x0000000199ec487c 34DE055D-8683-380A-9198-C3347211D13D + 104572
	15  Foundation                          0x0000000199ed2c68 34DE055D-8683-380A-9198-C3347211D13D + 162920
	16  Foundation                          0x0000000199ed26a8 34DE055D-8683-380A-9198-C3347211D13D + 161448
	17  Foundation                          0x0000000199ec6538 34DE055D-8683-380A-9198-C3347211D13D + 111928
	18  Foundation                          0x0000000199eff278 34DE055D-8683-380A-9198-C3347211D13D + 344696
	19  Foundation                          0x0000000199efecc0 34DE055D-8683-380A-9198-C3347211D13D + 343232
	20  Foundation                          0x0000000199efe730 34DE055D-8683-380A-9198-C3347211D13D + 341808
	21  libxpc.dylib                        0x00000002259c2b0c A46C2755-9586-33B8-9EA9-377F71175516 + 129804
	22  libxpc.dylib                        0x00000002259b5200 A46C2755-9586-33B8-9EA9-377F71175516 + 74240
	23  libdispatch.dylib                   0x00000001033ae094 _dispatch_client_callout3_a + 16
	24  libdispatch.dylib                   0x00000001033b23c8 _dispatch_mach_msg_async_reply_invoke + 388
	25  libdispatch.dylib                   0x000000010339c758 _dispatch_lane_serial_drain + 344
	26  libdispatch.dylib                   0x000000010339d5d8 _dispatch_lane_invoke + 484
	27  libdispatch.dylib                   0x00000001033a9894 _dispatch_root_queue_drain_deferred_wlh + 344
	28  libdispatch.dylib                   0x00000001033a8eb0 _dispatch_workloop_worker_thread + 580
	29  libsystem_pthread.dylib             0x0000000225958a0c _pthread_wqthread + 292
	30  libsystem_pthread.dylib             0x0000000225958aac start_wqthread + 8
)
(
	0   CoreFoundation                      0x000000019b34b228 7821F73C-378B-3A10-BE90-EF526B7DBA93 + 1155624
	1   libobjc.A.dylib                     0x00000001987e5abc objc_exception_throw + 88
	2   Foundation                          0x0000000199ed290c 34DE055D-8683-380A-9198-C3347211D13D + 162060
	3   Foundation                          0x0000000199ec6538 34DE055D-8683-380A-9198-C3347211D13D + 111928
	4   Foundation                          0x0000000199eff278 34DE055D-8683-380A-9198-C3347211D13D + 344696
	5   Foundation                          0x0000000199efecc0 34DE055D-8683-380A-9198-C3347211D13D + 343232
	6   Foundation                          0x0000000199efe730 34DE055D-8683-380A-9198-C3347211D13D + 341808
	7   libxpc.dylib                        0x00000002259c2b0c A46C2755-9586-33B8-9EA9-377F71175516 + 129804
	8   libxpc.dylib                        0x00000002259b5200 A46C2755-9586-33B8-9EA9-377F71175516 + 74240
	9   libdispatch.dylib                   0x00000001033ae094 _dispatch_client_callout3_a + 16
	10  libdispatch.dylib                   0x00000001033b23c8 _dispatch_mach_msg_async_reply_invoke + 388
	11  libdispatch.dylib                   0x000000010339c758 _dispatch_lane_serial_drain + 344
	12  libdispatch.dylib                   0x000000010339d5d8 _dispatch_lane_invoke + 484
	13  libdispatch.dylib                   0x00000001033a9894 _dispatch_root_queue_drain_deferred_wlh + 344
	14  libdispatch.dylib                   0x00000001033a8eb0 _dispatch_workloop_worker_thread + 580
	15  libsystem_pthread.dylib             0x0000000225958a0c _pthread_wqthread + 292
	16  libsystem_pthread.dylib             0x0000000225958aac start_wqthread + 8
)
```

I tried several other things, like using `MTRDevice` instead of `MTRBaseDevice`, building the command manually with `invokeCommand(…)`, or setting up the `MTRDeviceController` differently by initializing it with parameters, all with the same result.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

The one shipping in iOS 18.5 release (whatever that is…)

### Platform

darwin

bzbarsky-apple, 2025-05-16:
@Gernot this looks likely to be a bug in Home, not in the SDK, but can you give me an example of a command that _does_ work with this XPC setup for you, so I can compare the codepaths?

Gernot, 2025-05-16:
Sure! Here's one that does work, the device is an Eve Power Outlet with Matter:

```swift
let controller = MTRDeviceController.sharedController(
            withID: home.matterControllerID as NSCopying,
            xpcConnect: home.matterControllerXPCConnectBlock
        )

//...

let device = MTRBaseDevice(nodeID: NSNumber(value: nodeID), controller: controller)
let onOff = MTRBaseClusterOnOff(
    device: device,
    endpointID: NSNumber(value: 1),
    queue: .main
)

try await onOff?.on()
```

Thanks for looking into this!

closed 2025-05-20 by bzbarsky-apple

Gernot, 2025-05-26:
Thanks, I'll do. But that page also needs updating, it still refers to iOS 17.6 instead of 18.4, missing the relevant SHAs, and listing robot vacuums as "upcoming" which is relevant here.

bzbarsky-apple, 2025-05-30:
Yep, I'll see about getting that updated... like it says, "may be out of date".

Gernot, 2025-06-01:
Reported as FB17763037

Mostly copied from here, but with Sysdiagnoses. I also added the Home Logging Profile before submitting & reproducing, but when I added the Home Networking Profile the Home Logging Profile was gone. So I re-added the Hole Logging because I deemed it more important here. If that's not a bug, the instructions should make clear that only one of those profiles can be installed at a time, and it should make clear which one.